mantidproject/mantid #29970: Help > About should check user.properties file whether to enable usage data

DanielMurphy22, 2020-11-16:
related to #29956

### Expected behavior
First time set-up (AKA About MantidWorkbench window) sets the usage reporting tickbox according to the user's preference in the user.properties file.
### Actual behavior
It seems to always default to ticked, which is useful as it encourages users to report usage data.
This should be the default, but if a user edits the Mantid.user.porperties file separately, then it should be checked.
### Steps to reproduce the behavior
Set usage reporting to off by putting this line at the bottom of your Mantid.user.properties file: `usagereports.enabled = 0`
Open workbench and if the first time set-up window doesn't open then go to Help About MantidWorkbench
### Platforms affected
All but noticed on Windows SurfacePro

SilkeSchomann, 2020-12-02:
I've tried to reproduce the issue but cannot find the problem. The report usage box is ticked whenever usagereports.enabled = 1 in my mantid.user.properties file and unticked when usagereports.enabled = 0

DanielMurphy22, 2020-12-02:
@SilkeSchomann I'll take another look at this!

DanielMurphy22, 2020-12-02:
I am not able to reproduce this myself, so I'll get back to the user to get more info.

martyngigg, 2021-03-04:
@DanielMurphy22 Have we heard anything about this? If we haven't maybe we close it and we can reopen it if we hear more?

DanielMurphy22, 2021-03-04:
I think I am happy for this to be closed at the moment as it's probably very specific to the user's set-up. With https://github.com/mantidproject/mantid/pull/30731 the user shouldn't see the issue anyway.

closed 2021-03-04 by DanielMurphy22