mantidproject/mantid #29984: Indirect Reduction Save Custom Detector Grouping

robertapplin, 2020-11-18:
**Original reporter:** @SpencerHowells

### Description
Indirect Data Reduction needs the ability to save a custom grouping of detectors to a `.map` file in order for it to be easily loaded back into the interface.

This will require:
- Adding a button next to the Custom Grouping option that will save the grouping when clicked.
- Converting the custom group string into a `GroupingWorkspace`
- Modification of the `SaveDetectorsGrouping` algorithm to allow the saving of `.map` files.

### Platforms affected
All

robertapplin, 2020-12-07:
@SpencerHowells does indirect ever use `.xml` files for detector grouping files, or is it only ever `.map`?

SpencerHowells, 2020-12-07:
@robertapplin the only feedback/comments I’ve had is from some of our scientists is that they use the .map because a text file is easier to use. We also think that for that reason it would be best for the users.

closed 2020-12-16 by martyngigg