mantidproject/mantid #30008: Pearl absorption correction process not tolerant of custom datasets with different number of histograms

DannyHindson, 2020-11-20:
Chris Ridley has requested a small change to the Pearl powder diffraction scripts. He would like to be able to run a create_vanadium process on some runs that are non-standard. These runs have a non-standard number of histograms in them and when Mantid gets to the point of dividing the datasets by the absorption correction workspace it raises an error.

Further detail below
```
Hi Danny – thanks for looking into this. I think the `divide-by’ switch may be the best option – it gives us the option to test cards like this again easily if we want to. 

Does it throw any errors if the grouping file doesn’t reference them? I.e, if we focus the rest of the detectors as normal, can it easily just ignore and then discard the extra spectra by default until we change the grouping file?

Best Wishes,
Chris

From: "Hindson, Danny (STFC,RAL,ISIS)" <danny.hindson@stfc.ac.uk>
Date: Tuesday, 17 November 2020 at 17:27
To: "Ridley, Christopher (STFC,RAL,ISIS)" <christopher.ridley@stfc.ac.uk>, "Funnell, Nick (STFC,RAL,ISIS)" <nick.funnell@stfc.ac.uk>
Subject: RE: PEARL routines

Hi Chris,
 
I’ve had a look at this and yes I can recreate the problem. The absorption corrections are applied by dividing the vanadium data for cycle 19_1 (which will have 1688 histograms) by the workspace “pearl_absorp_sphere_10mm_newinst2_long.nxs” which only has 1064 histograms. I get this error:
 
 
 
How should the absorption be applied if the absorption file has different number of histograms to the vanadium data? The counts in the spectra numbered 1065 upwards in the 19_1 data look like they’re all zero so can these “extra” spectra just be ignored or should they be removed? We could use a switch on the Mantid Divide algorithm called “AllowDifferentNumberOfSpectra”. This is off at the moment but if we turn it on then the Divide process would match the spectra in the two workspaces being divided based on detector id. Any non-matching spectra in the “left hand” workspace just don’t get divided by anything but are left in the output from Divide.
 
Thanks
Danny
 
From: Ridley, Christopher (STFC,RAL,ISIS) <christopher.ridley@stfc.ac.uk> 
Sent: 12 November 2020 16:50
To: Funnell, Nick (STFC,RAL,ISIS) <nick.funnell@stfc.ac.uk>; Hindson, Danny (STFC,RAL,ISIS) <danny.hindson@stfc.ac.uk>
Subject: PEARL routines
 
Hey Danny, Nick – I have stumbled across a minor issue with the new focussing routines. 
 
For a couple of cycles last year we performed some tests for the new HRPD detector cards. They split the signal from a set of our detectors, and simulated new histograms from them. I think this is giving me some issues for creating the vanadium for e.g cycle 19_1. It appears to run with `do_absorb_corrections=False’, but when set to true the different number of histograms throws an error on divide . We got around this in the old scripts by blanket cropping all runs to match the required 1064 histograms, rather than 1688. 
 
Would you mind checking that this is something you can repeat, and whether we can fix?
 
Best Wishes,
Chris

DanielMurphy22, 2021-02-12:
@DannyHindson as you mentioned, if I simply set line 38 in pearl_args.py to 
```
 van_ws = mantid.Divide(LHSWorkspace=van_ws, RHSWorkspace=absorb_ws, OutputWorkspace=van_ws, AllowDifferentNumberSpectra=True)
```
then the create_vandium runs fine. I assume this is the result is correct but not sure how to check.

As to what Chris mentioned:

> Does it throw any errors if the grouping file doesn’t reference them? I.e, if we focus the rest of the detectors as normal, can it easily just ignore and then discard the extra spectra by default until we change the grouping file?

There are no errors output with this change, so I assume that the extra detectors are all ignored. The extra detectors were all `DetID = 0, SpectrumNum = -1, Monitor = n/a` so this is probably a safe assumption.

- Is it okay to set this `Divide(...AllowDifferentNumberSpectra=True)` in all cases?

DannyHindson, 2021-02-12:
> @DannyHindson as you mentioned, if I simply set line 38 in pearl_args.py to
>
> ```
>  van_ws = mantid.Divide(LHSWorkspace=van_ws, RHSWorkspace=absorb_ws, OutputWorkspace=van_ws, [...]

The code change you've made is within a process to apply absorption corrections to the Vanadium data. A little bit later on in the data reduction process (in calibrate.py) the data gets focussed - basically the data for the detectors in each bank is aggregated. The focussing step is driven by a grouping file that tells Mantid which detectors are in which bank. If there are some extra detectors that aren't in the grouping file they should just get dropped and from there on the dataset will be the same as with a normal set up - so the "HRPD" funniness will have gone. The fact you're not seeing any errors suggests this is happening OK.
> 
> * Is it okay to set this `Divide(...AllowDifferentNumberSpectra=True)` in all cases?

Yes that's fine. The code change you've made is in a Pearl-specific unit so won't affect any other instruments. And I think that for normal Pearl runs the number of spectra in the Vanadium data will always equal the number of spectra in the absorption correction dataset so the new setting won't do anything unless Chris is focussing one of these unusual hybrid HRPD\Pearl runs.

DanielMurphy22, 2021-02-12:
Thanks, that makes sense! I think getting them to test the nightly with these changes is a good idea!

PR #30693 (2021-02-26): Allow custom dataset for PEARL create_vanadium
**Report to:** Chris Ridley

**Description of work.**

Absorption correction division now allows for non-standard
datasets with a different number of spectra.

**To test:**

Using the `DivideProblem_Pearl.zip` folder under `babylon/Public/Daniel Murphy/` run the Test_Divide_Pearl+HRPD.py script which runs a `create_vanadium` process with `do_absorb_corrections=True` for a hybrid Pearl/HRPD detector set-up (see issue).

The process should run correctly, producing the PEARL and spline workspace groups, without the errors from the initial issue.

Fix [... cut]
closed 2021-02-26 by martyngigg