mantidproject/mantid #30014: 3D axis limits on Linux not handled properly

DanielMurphy22, 2020-11-20:
Found when creating #29926  

### Expected behavior
Double-click menus to edit axis limits on 3D plots work smoothly on Linux
### Actual behavior
Works fine on Windows and Mac, but on Linux (and IDAaaS) it doesn't work correctly
as the matplotlib version is on 2.2.3. (MAc and Windows are on 3.1.3 and 3.1.2 respectively)
### Steps to reproduce the behavior
Run: 
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

ws = CreateSampleWorkspace(Function = 'Exp Decay')
fig, ax = plt.subplots(subplot_kw={'projection':'mantid3d'})
ax.plot_surface(ws)
plt.show()
```
then double-click on the z axis values and enter `Max = 0.2`

error message: `'MantidAxes3DSubplot' object has no attribute 'get_zaxis'`
### Platforms affected
Linux / IDAaaS Workbench 5.1.1 and nightly

DavidFair, 2022-01-06:
Unstaling, as this was found in #33187

martyngigg, 2022-01-13:
Fixed by #33259

PR #33259 (2022-01-12): Fix not being able to set z axis limits on linux
**Description of work.**

The `get_zaxis()` method did not exist in the older version of mpl used on linux (incl. IDaaS). This simply accesses the value of the attribute directly which exists in the same form in both versions we use.

**To test:**
Ideally the functional testing should be done on both windows/mac **and** linux to check that the functionality has not changed for the former and is fixed for the latter. 

1. Run:
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
impor [... cut]
closed 2022-01-13 by martyngigg