mantidproject/mantid #30017: Manual Testing Workbench Group 1

DanielMurphy22, 2020-11-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/Core/Core.html

Matt-Cumber, 2020-12-10:
Will be testing on Mac

Matt-Cumber, 2020-12-10:
Not sure why but when opening an interface it always forces the top of the interface to be at the top of my screen, and I can't re-adjust the height which is problematic when interfaces have buttons at the bottom of them. As an example:
[image]

And in full screen:
[image]

Matt-Cumber, 2020-12-10:
Simply visual and for consistency have Axis as the title of this group box like limits has a title in plot settings:
[image]

Matt-Cumber, 2020-12-10:
When you change error every in plot settings for error bars, when you change the value the plot updates but the value in the line edit is reset to 1 on the interface, yet if you press apply again nothing changes (i.e. the value is successfully stored on the backend but reset on the GUI)

Matt-Cumber, 2020-12-10:
Markers on plots not saved correctly:
[image]

Created marker 0 and 1 then saved project then reload, they appear stuck at bottom left, and added new marker also called 0

Matt-Cumber, 2020-12-10:
Changing plot to waterfall and changing axis from context menu only changes one spectrum:
[image]

(Ignore the markers from previous comment)
Plot multiple spectra, right-click change to waterfall, then change axis scale

Matt-Cumber, 2020-12-11:
Bug:
1. Load in some data
2. Plot a spectrum (I was plotting 1)
3. Go to plot settings and try to remove the only curve
4. Error reporter

Matt-Cumber, 2021-01-18:
Closing this issue now as have conducted testing on a recent nightly

closed 2021-01-18 by Matt-Cumber