mantidproject/mantid #30021: Manual Testing Muon

DanielMurphy22, 2020-11-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See instructions at https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
http://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

Harrietbrown, 2020-12-10:
Starting with http://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

Documentation issues:
MuonTesting refers to tabs that have been renamed (`Data Analysis Tab`) or removed (`Settings`)
The global checkbox is dissabled unless `Simultaneous fit over` is checked
there are numerous things like this in `MuonTesting` that needs someone who knows the new layout to point out what is what.

When I have made issues for these bugs i will attach them here

1. open muon analysis
2. load EMU51341
3. in the fit tab add a flat background
4. add DynamicKuboToyabe. this causes a crash

1. open muon analysis
2. load EMU51341-3
3. in the fit tab add DynamicKuboToyabe
4. check Simultaneous fit over, change combo box from `Runs` to `Group/Pair` this causes a crash

1. open muon analysis
2. load EMU20889-20900 and check co-add
3. go to the fitting tab and check `Simultaneous fit over` this raises an index error

AnthonyLim23, 2020-12-11:
I have checked, it does not fall over in the generic fitting

AnthonyLim23, 2020-12-11:
> Starting with http://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html
>
> Documentation issues:
> MuonTesting refers to tabs that have been renamed (`Data Analysis Tab`) or removed [...]
looks like a muon specific bug. I suspect the one below is related.

> --
> 
> 1. open muon analysis
> 2. load EMU51341-3
> 3. in the fit tab add DynamicKuboToyabe
> 4. check Simultaneous fit over, change combo box from `Runs` to `Group/Pair` this causes a crash
> 
> --
> 

> 1. open muon analysis
> 2. load EMU20889-20900 and check co-add
> 3. go to the fitting tab and check `Simultaneous fit over` this raises an index error
> 
> --
Looks like the simultaneous fit over does not pick up the co-added run correctly. 

> In the Superconducting Copper Test it calls for the results workspace to be converted to a matrix workspace, but the output result worksapce can not be converted.
> 
> --
> 
> 1. open muon analysis.
> 2. load MUSR62260.
> 3. go to fitting tab.
> 4. add a flat background.
> 5. check `TF Asymmetry Mode`.
> 6. check `Simultaneous fit over`.
> 7. change combo box from `run` to `Group/Pair` This produces an error.
To be consistent with the rest of the behaviour for TF Asymmetry mode we should not allow the user to change the simultaneous fit options. Part of the problem is the function form changing. We can do this by doing a check when things are changed and if TF asymmetry mode is on we can give a warning and reset it

Harrietbrown, 2020-12-11:
https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_1_general.html#muon-analysis-testguide-1-general-ref

The load for PSI is disabled, what is the current status of that?
MUT is no longer an instrument listed
the rebinningis now in the home tab, not settings.

https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_2_fitting.html

"Click and drag blue lines on graph, check start/end times are updated." the blue lines are not present in the miniplot

--

1. open muon analysis
2. load `EMU20918-20920`, the plot window has a y range down to 1,000,000. far outside of the ragne of the data

--

1. open muon analysis
2. load `EMU20918-20920`
3. in fitting add a flat background
4. fit, it won;t be a good fit but that is expected
5. check co-add
6. click fit, you get an error that the function is empty depite having a background.

--

Label box is now called function name

--

`Groups/Pairs to fit` is not in the fitting tab, and there is not custom option in the `Simultaneous fit over` options

--

1. open Muon analysis
2. Load `EMU62261`
3. in grouping tick analyse for `fwd` and `bwd`
4. in fitting add GausOsc
5. check `Simultaneous fit over` 
6. change runs to group/pair and select fwd.
7. tick `TF Asymmetry mode` This causes an error

Harrietbrown, 2020-12-11:
https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_3_results.html

When you co-add runs the Sequential Fitting tab uses the run number for only the first run. overwriting single fits

--

Results table can only have the result from a snigle fit, so combining ones for co-add and single fits can't be done.

--

This whole set of tests need to be redone for the current results tab, but there are no crashes or errors hwen outputting tables.

https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_4_FDAG.html

In test 3 it says that when you put in a PhaseQuad a new line in the table, `axes`, should appear it does not, so i cannot change it for later steps.

The naming and organisaton of workspaces has changes so I do not know what exactly I am looking for for the MaxEnt output.

AnthonyLim23, 2020-12-11:
> https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_1_general.html#muon-analysis-testguide-1-general-ref
> 
> The load for PSI is disabled, what is the current status of that?

Just had a look with the new GUI. If you set the instrument to PSI, it should work ok

> MUT is no longer an instrument listed

I dont think its used

> the rebinningis now in the home tab, not settings.
>

Another doc update ;)
> https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_2_fitting.html
> 
> "Click and drag blue lines on graph, check start/end times are updated." the blue lines are not present in the miniplot
>

Not implemented yet

> --
> 
> 1. open muon analysis
> 2. load `EMU20918-20920`, the plot window has a y range down to 1,000,000. far outside of the ragne of the data
> 

we dont do any checks on the data range

> 
> 1. open muon analysis
> 2. load `EMU20918-20920`
> 3. in fitting add a flat background
> 4. fit, it won;t be a good fit but that is expected
> 5. check co-add
> 6. click fit, you get an error that the function is empty depite having a background.

I think this is co-add not working again

> --
> 
> Label box is now called function name
> 

doc update

> 
> `Groups/Pairs to fit` is not in the fitting tab, and there is not custom option in the `Simultaneous fit over` options
> 

another doc update

> --
> 
> 1. open Muon analysis
> 2. Load `EMU62261`
> 3. in grouping tick analyse for `fwd` and `bwd`
> 4. in fitting add GausOsc
> 5. check `Simultaneous fit over`
> 6. change runs to group/pair and select fwd.
> 7. tick `TF Asymmetry mode` This causes an error

this might be the data. This should only be done with transverse data (e.g. MUSR 62260)

AnthonyLim23, 2020-12-11:
> https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_3_results.html
> 
> When you co-add runs the Sequential Fitting tab uses the run number for only the first run. overwriting single fits
> 

co-add does not work

> --
>
> Results table can only have the result from a snigle fit, so combining ones for co-add and single fits can't be done.
>
> --
>
> This whole set of tests need to be redone for the current results tab, [...]

I think this ok

> The naming and organisaton of workspaces has changes so I do not know what exactly I am looking for for the MaxEnt output.

The MaxEnt should be plotted. In the ADS it will have MaxEnt in the name

Harrietbrown, 2020-12-11:
> Just had a look with the new GUI. If you set the instrument to PSI, it should work ok

this is what I mean when I say load is disabled

[image]

AnthonyLim23, 2020-12-11:
ah yes, thats because we don't know what the current run is and requires the user to have local copies.

Harrietbrown, 2020-12-11:
OK that's fine then

Harrietbrown, 2020-12-11:
#30147
#30148
#30149
#30150

Harrietbrown, 2021-01-12:
https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html
In the Superconducting Copper Test it says to try using the > button, this gets rid of the multiple load so the output following these instructions does not result in the correct output. this is a documentation issue.

in the HIFI Transverse Field Simultaneous Fitting test the fit results in a Chi squared of 284.9 but still lists it as having succeeded. THIS MAY BE SIGNIFICANT.
the line `Enter values for each run in the table as from 0.1 to 1.1 in steps of 0.1` should go up to `1.2`, not `1.1` This is a documentation issue

Harrietbrown, 2021-01-12:
1. Change instrument to EMU 
2. load current
3. the plot window does not re scale to the loaded run

Harrietbrown, 2021-01-12:
for all instruments apart from EMU and MUSR I am getting the `Cannot find file : \\<instrument>\data\autosave.run` when I try to load current run. I can still load individual runs.

Harrietbrown, 2021-01-12:
in https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_1_general.html#test-3-basic-fitting-in-the-gui it referecnes the wrong test to use the settings from `Start with the same data loaded and rebin settings from Test 2: Sequential Fit`

AnthonyLim23, 2021-01-12:
> https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html
> In the Superconducting Copper Test it says to try using the > button, this gets rid of the multiple load so the output following these instructions does not result in the correct output. this is a documentation issue.
> 
> in the HIFI Transverse Field Simultaneous Fitting test the fit results in a Chi squared of 284.9 but still lists it as having succeeded. the chisq was still high so this may be ok but should be looked into.
> the line `Enter values for each run in the table as from 0.1 to 1.1 in steps of 0.1` should go up to `1.2`, not `1.1` This is a documentation issue

it should start at 0 not 0.1, thats probably my mistake.

AnthonyLim23, 2021-01-12:
For fitting success just means that it is confident it has found a minima, that does not mean it is the global minima. Hence, the large chi-squared value.

AnthonyLim23, 2021-01-12:
> for all instruments apart from EMU and MUSR I am getting the `Cannot find file : \\<instrument>\data\autosave.run` when I try to load current run. I can still load individual runs.

I have just looked on version 4.2, using both the new and old GUI's. It has the same problem. So I think it is an issue with the data storage.

Harrietbrown, 2021-01-12:
as part of https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_2_fitting.html#test-2-sequential-fit
When loading EMU20918-20920 only 20918 is being added to the sequential fitting tab.

Harrietbrown, 2021-01-13:
In FDA
These tests were not updated in #30245 so some steps are not accurate.
when FDA does a transform the outputs are put into a group workspace, not a snigle workspace with multiple spectra.
in https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_4_FDAG.html#test-3-phasequad it says
```
Change the workspace to “PhaseQuad”
You should not be able to select the workspace for the imaginary part, but it is possible to turn the imaginary component on and off.
```
The imaginary component can be selected. and there is also no `generate new phase table` option that is also mentioned in the test.

MaxEnt produces a sadmantid when attempted to be calculated, an issue will be made for this.

AnthonyLim23, 2021-01-14:
we have changed phasequad a bit this release, thats a good spot.

I just tried the maxent on my dev build and it worked. I remember fixing a crashin maxent just before the holiday.

closed 2021-02-04 by martyngigg