mantidproject/mantid #30025: Manual Testing Engineering Diffraction

DanielMurphy22, 2020-11-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow: http://docs.mantidproject.org/nightly/interfaces/Engineering%20Diffraction%20Test%20Guide.html

DannyHindson, 2021-01-12:
URL for testing guide seems to have changed to:
https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html

DannyHindson, 2021-01-12:
@RichardWaiteSTFC - I've completed the manual testing on the engineering diffraction UI. There are two things need changing I think and then a few lower priority issues which probably don't need doing for the release

1) the testing guide needs some updates. I've tested the Fitting tab anyway but I think we should fill in this gap now so it's ready for next time:
- add in some content for the Fitting tab and remove the "work in progress" comment about the fitting tab from Overview
- Test 1: .abc files are produced now rather than .dat
- Bullet number 4 in "Test 3" needs indenting
2) the default path in the "Focused Run Files" on the fitting tab was missing a "User" level if the last focus had an RB number specified. This gave a "file not found" message when hitting Load

Here are the low priority issues:
1) the undocked plot window on fitting tab - since redocking requires clicking on header, it might be useful if the header was a diff colour. Not a big deal
2) I think you can load a mix of d spacing and TOF at the moment and plot them all on same plot? Can this be restricted somehow so the loaded workspaces are either all TOF or all d spacing?
3) the Sequential Fit button is enabled when you add a function to the fit property browser but not disabled if you remove a function by right clicking on the function and choosing "Remove". Pressing Sequential Fit after doing this causes Mantid to crash. The button is disabled if you do the "Clear Model" action so v likely a feature of the fit property browser code that we rely on

Finally, one question. The Sequential Fit button doesn't give a v strong indication when it's finished. As far as I can tell the only change is that the xxx_fits workspace appears in the main workbench window. If that's correct, is there any way of giving the user a more clear "finished" indication? Even adding a message to the log would be useful.

DannyHindson, 2021-01-12:
Have created this issue:
https://github.com/mantidproject/mantid/issues/30332

DannyHindson, 2021-01-12:
Closing because testing complete. My testing was done on Windows

closed 2021-01-12 by DannyHindson