mantidproject/mantid #30030: Manual Testing Live Data ISIS

DanielMurphy22, 2020-11-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html

gemmaguest, 2021-01-18:
Testing on Windows. I got this error on closing sliceviewer, which had been opened on a live data workspace. This was for the `Live Histogram Data` test in the instructions.
```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\sliceviewer\presenter.py", line 381, in refresh_view
    self.view.data_view.image_info_widget.setWorkspace(self.model._get_ws())
RuntimeError: wrapped C/C++ object of type ImageInfoWidget has been deleted
```

gemmaguest, 2021-01-19:
I found an issue with running the fake DAE algorithms in the interpreter: #30407

gemmaguest, 2021-01-19:
I've raised #30406 for the sliceviewer regression. No other issues found.

closed 2021-01-19 by gemmaguest