mantidproject/mantid #30033: Manual Testing Project Recovery

DanielMurphy22, 2020-11-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

Matt-Cumber, 2021-01-12:
Just as a note in test 3 it references `/method_test_r.csv` twice instead of `/method_test.csv` and `/method_test_r.csv`

Matt-Cumber, 2021-01-12:
Also could do with updating as has some mantid plot images

Matt-Cumber, 2021-01-12:
No issues found directly relating to project recovery so closing issue

closed 2021-01-12 by Matt-Cumber