mantidproject/mantid #30034: Manual Testing Error Reporter

DanielMurphy22, 2020-11-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ErrorReporterTesting.html

AnthonyLim23, 2021-01-04:
I have done the tests. The doc does need updating. @DanielMurphy22 did the emails come through (should have the public, private and private2)

DanielMurphy22, 2021-01-07:
I think that should be fine! Is the only thing that needs updating in the testing doc to remove references to MantidPlot?

AnthonyLim23, 2021-01-07:
yep, I just used workbench instead

closed 2021-01-12 by AnthonyLim23