mantidproject/mantid #30035: Manual Testing Sliceviewer

DanielMurphy22, 2020-11-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Test for both MDEvent and MDHisto workspaces: Briefly I tested for both MDEvent and MDHisto workspaces:
[ ] 1. Viewing 2D,3D and 4D MD data - (changing bins and sliders, color limits, colormap etc.)
[ ] 2. Panning, home, Zooming/Dynamic rebinning - both by selecting region with mouse and scrolling
[ ] 3. Toggling tracking of cursor
[ ] 4. Toggle gridlines
[ ] 5. Experiment with line plots and box integration (disabled fro non-orthogonal data)
[ ] 6. Changing viewing axes
[ ] 7. Peak overlay (multiple peaks workspaces, integrated peaks, sorting table, zooming on peaks, changing axes with peaks shown, peak overlay should not be shown for 2D data)
[ ] 8. Repeat (1-7) for data with non-orthogonal axes
[ ] 9. Play around with window - resizing, adjusting peak table size
[ ] 10. Saving plot
[ ] 11. Delete/rename underlying workspaces (at present sliceviewer does not update)

DanielMurphy22, 2021-01-07:
While doing the Core testing on IDAaaS, I had a quick look at Sliceviewer and thought it was worth checking these with @RichardWaiteSTFC.
- Firstly a question, do you think users should be able to open more than 1 instance of sliceviewer on one workspace?
- Secondly, I ran the script to get the non_ortho data (from my own instructions 😆) and after opening this `histoWS_3D` data in Sliceviewer, if I zoom in or pan across the data or change the displayed axes then I get this error:
Traceback (most recent call last):
```
 File "/opt/mantidnightly/lib/mantidqt/widgets/sliceviewer/view.py", line 403, in on_data_limits_changed
   self.presenter.data_limits_changed()
 File "/opt/mantidnightly/lib/mantidqt/widgets/sliceviewer/presenter.py", line 190, in data_limits_changed
   self.new_plot()  # automatically uses current display limits
 File "/opt/mantidnightly/lib/mantidqt/widgets/sliceviewer/presenter.py", line 83, in new_plot_MDH
   self.new_plot_MDE()
 File "/opt/mantidnightly/lib/mantidqt/widgets/sliceviewer/presenter.py", line 109, in new_plot_MDE
   limits=limits))
 File "/opt/mantidnightly/lib/mantidqt/widgets/sliceviewer/model.py", line 159, in get_ws_MDE
   return BinMD(InputWorkspace=workspace, OutputWorkspace=self._rebinned_name, **params)
 File "/opt/mantidnightly/lib/mantid/simpleapi.py", line 1036, in __call__
   algm.execute()
RuntimeError: SlicingAlgorithm::createTransform(): Cannot propagate a transformation if the number of dimensions has changed.
```

RichardWaiteSTFC, 2021-01-11:
@DanielMurphy22
(1) I think this is OK as long as the operation in one window do not corrupt another - this seems to work fine, I have tried rebinning, zooming etc. and the windows operate entirely independently (despite there only being one rebinned workspace).
(2) I haven't been able to reproduce this fault using WISH data - how did you create the MD object?

RichardWaiteSTFC, 2021-01-11:
I can reproduce the fault in issue #30307 for 3D data as well - it only seems to happen when swapping the current viewing axes, i.e. if you select an different axis to view the cursor tracking resets correctly.

RichardWaiteSTFC, 2021-01-11:
Have created an issue due to exception when changing a peaks workspace that has been overlaid #30308

RichardWaiteSTFC, 2021-01-11:
No other new issues to come out of testing, but I wanted to draw attention to two existing issues #29105 and #29087 that are very important for the single-crystal diffraction workflow

closed 2021-01-11 by RichardWaiteSTFC