mantidproject/mantid #30041: LoadElementalAnalysisData fails if directory contains a . character

martyngigg, 2020-11-21:
### Expected behavior

`LoadElementalAnalysisTest` & `LoadPartialElementalAnalysisTest` should pass for all valid directories.

### Actual behavior

The [Ubuntu system tests](https://builds.mantidproject.org/job/master_systemtests-ubuntu-18.04/402/) have the data in a directory like `/jenkins_workdir/workspace/master_systemtests-ubuntu-18.04/ExternalData/Testing/Data/SystemTest`. When [LoadElementalAnalysisData.py](https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/LoadElementalAnalysisData.py#L83) attempts to deconstruct the filename it does so with the whole path and the `.` in the directory gives an extra item in the list and the `TYPE_KEYS` lookup fails.

### Steps to reproduce the behavior

I was able to reproduce this by creating a temporary directory with a `.` in it:

```
cd <mantid_build_dir>
mkdir temp.dir
```

copying all of the `ral*.dat` files from `ExternalData/Testing/Data/SystemTest` into it and running the tests as:

(on Windows)
```
systemtest -C BUILD_CONFIG -d %CD%\temp.dir -R LoadElementalAnalysis
```
where BUILD_CONFIG is the build type last built.

(on Linux/mac)
```
systemtest -d ${PWD}\temp.dir -R LoadElementalAnalysis
```

### Platforms affected
All

SpencerHowells, 2020-11-21:
@martyngigg There is also a problem with a space. On Windows it will occur if you use the default folder ‘My PC’.

martyngigg, 2020-11-23:
@SpencerHowells Thanks for finding that.

PR #30050 (2020-11-24): Use filename not path to generate variables in LoadElementalAnalysisData
**Description of work.**

Changes to LoadElementalAnalysisData Algorithm to fix issues with directories containing `.` 

**To test:**

Replicate test from #30041 

Fixes #30041  

This does not require release notes because LoadElementalAnalysisData Algorithm already has documentation and this change does not affect that

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
 [... cut]
closed 2020-11-24 by gvardany