mantidproject/mantid #30048: Replace EnggCalibrate/Full algorithms with PDCalibration

RichardWaiteSTFC, 2020-11-23:
PDCalibration is an existing C++ implementation that is widely used (and well tested) that performs a large chunk of the Calibration currently done in python in the EnggCalibrate and EnggCalibrateFull algorithms. The output of PDCalibrate is also compatible with other algorithms for unit conversion which should replace usages of ConvertUnits in the Engineering UI code.

DannyHindson, 2021-03-01:
@RichardWaiteSTFC - I've just moved this issue that Tom's working on into the new sprint. I think you said you had a python script that describes the required calibration workflow. Please could you zip it up and attach it here?

RichardWaiteSTFC, 2021-03-30:
Thought I'd link this here so we don't forget - #30796 suggests small change to proposed workflow in script (i.e. don't use EditInstrumentGeometry)

closed 2021-06-16 by gemmaguest