mantidproject/mantid #30062: Fix problems in ISISIndirectBayesTest.QLWidthTest system test

DannyHindson, 2020-11-24:
The ISISIndirectBayesTest system test suite (Testing\SystemTests\tests\framework\ISISIndirectBayesTest.py) is run on Windows only as part of the Windows nightly build. It's not run as part of the PR Jenkins processes because the system tests there are run on RHEL only.

The system test is randomly failing in the Windows nightly build process which means the nightly build installer isn't available on the Mantid website every day. You can see the failures on this screen:

https://builds.mantidproject.org/view/Master%20Pipeline/job/master_systemtests-windows/

If you drill into the highlighted jobs you can see this system test has failed:

[image]

I started investigating this and found a couple of problems but haven't had time to finish this off. The problems I found are these:

a) Most of the tests call _cleanup_files at the end and there is a warning raised if any of the files that the test attempts to delete are not found. The files that are specified in the call to _cleanup_files are wrong in some cases meaning the cleanup doesn't actually happen. Specifically - QLRestNormTest and QLWidthTest both try to delete files with _QLd in the file name but the tests actually produce files with _QLr in their name
b) I think QLresTest, QLResNormTest and QLWidthTest all generate intermediate files with the same name so there's a risk they could interfere with each other if the tests get run in more than one thread. I'm not sure what flexibility we have to change the names per test. It might be possible to work this out by looking inside BayesQuasi.py and see what logic is generating the names.

Based on some quick tests I was still seeing the QLWidthTest failing even when I ran it on its own which suggests that fixing a) and b) won't totally solve the problem. But fixing this will clear away some distractions and help us move onto fixing any further issues.

Couple of useful things:
- I think the intermediate files get written into the build\Testing\SystemTests\scripts folder so you can run the tests locally and see the problems I'm talking about.
- You can run this system test by running the following from your build folder:
systemtest.bat -C Debug -R ISISIndirectBayesTest.QLWidthTest
...or this will run the whole set of tests in the ISISIndirectBayesTest.py unit:
systemtest.bat -C Debug -R ISISIndirectBayesTest

SpencerHowells, 2020-11-24:
@DannyHindson @SilkeSchomann the Bayes programs run FORTRAN modules. It only works on Windows and is unavailable on Linux & Mac at the moment.

SpencerHowells, 2020-11-25:
@DannyHindson @SilkeSchomann The resolution ws can be *_res (1 histogram) or *_red (many histograms). For the former it runs QLr; for the latter it runs QLd. So there is a parameter in the Py script that keeps track of this.

SpencerHowells, 2020-11-25:
@DannyHindson @SilkeSchomann QLwidth works with output from both QLres & QLdat. The ws format is the same.

martyngigg, 2021-01-04:
Fixed by #30165

closed 2021-01-04 by martyngigg