mantidproject/mantid #30071: Ragged workspace double click crash

AnthonyLim23, 2020-11-25:
If you create a ragged workspace then double click on it in workbench, Mantid will give a terminate message box.

martyngigg, 2020-11-26:
Hmm. I thought, @robertapplin fixed this in #29589? Is it a different error?

robertapplin, 2020-11-26:
I think this is a different error than was fixed in #29862 . In #29862 it was erroring when right clicking the ragged workspace and selecting colorfill plot

AnthonyLim23, 2020-11-26:
It does seem to be different

martyngigg, 2020-11-26:
My apologies, I got the pull request link wrong. I fixed the issue link above to refer to the one where we did the padding for the display.
@AnthonyLim23 Do you expect someone in your team to be able to take a look at this?

AnthonyLim23, 2020-11-26:
Not got anything planned yet. @robertapplin do you think you could fix it?

robertapplin, 2020-11-26:
Yep I'll have a look

robertapplin, 2020-11-27:
@AnthonyLim23 @martyngigg What is the expected behaviour for a matrixworkspace? To plot colorfill?

AnthonyLim23, 2020-11-27:
I thinks its plot

martyngigg, 2020-11-27:
Was it not to bring up the dialog to ask what to plot? We don't know what they want to do yet: 1d, 2d plot but the dialog has all of the options.

robertapplin, 2020-11-27:
Yes I think your right. We want the dialog which asks what to plot

PR #30079 (2020-12-16): Fix Double Click of Ragged workspaces
**Description of work.**
This PR fixes a bug where an error message appears after double clicking a ragged workspace in the ADS.

The problem was caused by the `blocksize` method (which is not compatible with ragged workspaces). The ongoing issues with ragged workspaces has shown there are several missing methods in the MatrixWorkspace API. @martyngigg has suggested adding:

- `getNumberBins(std::size_t index)` can now be used to find the number of bins for a given histogram index.
- `getMaxNumberBins()` will get the maximum number of bins in a workspace ( [... cut]
closed 2020-12-16 by gemmaguest