mantidproject/mantid #30098: Workbench won't launch on Linux with MPL 3.3

DanielMurphy22, 2020-12-01:
**Original reporter:** Oliver via [Forum](https://forum.mantidproject.org/t/mantid-5-1-crash-at-start-on-unbuntu-18-04/703)

On Mac and Windows, Mantid currently uses MPL 3.3 (Slightly different versions)

On Linux, by default we use MPL 2.2.3, but it can work for users with higher MPL versions.
However, MPL 3.3 and higher doesn't seem to work for Linux. This will become a problem when we start to have more Ubuntu 20.04 users.
### Steps to reproduce the behavior
- On Linux, have a mantid package installed
- On a command line run `pip3 install matplotlib==3.3.2`
- try to open Workbench
### Platforms affected
Linux

StephenSmith25, 2021-02-18:
I have investigated this and found a few issues:

Firstly, workbench isn't opening as since Matplotlib 3.3ish, the backend tries to copy over the method signature:
Before: https://github.com/matplotlib/matplotlib/blob/782f6ef56acca21a4e1057ef3e9aecf5e22e88a2/lib/matplotlib/pyplot.py#L242
Now: https://github.com/matplotlib/matplotlib/blob/422e5374f04423bac416ec6c9b8afa2d619f5285/lib/matplotlib/pyplot.py#L295
I think it appears the QAppThread wrapper means there is no signature, a possible fix is:

```
show = QAppThreadCall(show_impl)
show.__signature__ = inspect.signature(show_impl)
```

Fixing this leads to another issue that in Matlotlib 3.3 the API for toolbars has changed. In Mantid we are subclasssing a subclass of the toolbars, where we override the method `_init_toolbar()` in order to construct our custom toolbar.  This method in 3.3 is deprecated and now it expects the toolbar to be fully initialized in its constructor.  There's more issues in here as our tool dictionary isn't comfortant with how matplotlib expects the tools to be defined...

Another warning (which pops up when you open workbench) is that we are accessing the matplotlib colormaps directly, which is now is now a private dictionary in 3.3.

Given there's quite a few issues, I think this will require quite a lot of changes to work. And needs more planning to decide when the correct time to upgrade is.

Pasarus, 2021-04-01:
Looking at this, there are a myriad of version mismatches in the dependencies including Numpy, Scipy, and Matplotlib.
To deal with these issues:
- Update matplotlib to 3.3.2
- Update numpy to 1.19.5
- Update scipy to 1.1.0
This is before dealing with the other issues.

Pasarus, 2021-04-08:
Due to some further investigation, this issue will not be worked on in this sprint, due to it requiring significant internal changes with potentially wide ranging effects so close to a release cycle. Since none of the systems we currently support or will be shipping the release with have Matplotlib 3.3 as a requirement or a default package, this is not something that has to be tackled right now.

The changes made so far that are necessary are:
- Remove the `is_pyqt5()` function usage and the pyqt4 part of it, as we no longer support it and is_pyqt5 is now deprecated
- Replace usage of `cmap_d.keys()` and replace with `pyplot.colormaps`
- The Matplotlib navigation qt5 toolbar now uses toolitems in it's internal API so we need to change our internal variable name of toolbar's toolitems to something like that `custom_toolitems` just so it's different as they have different APIs.
- Matplotlib now needs the figure managers to have a function called `_set_new_active_manager` to have the same API.
- Replace the QAppThreadCalls in `backend_workbench.py` with:
```
draw_if_interactive = QAppThreadCall(draw_if_interactive_impl)
draw_if_interactive.__signature__ = inspect.signature(draw_if_interactive_impl)
show = QAppThreadCall(show_impl)
show.__signature__ = inspect.signature(show_impl)
```

PR #31949 (2021-07-12): Support for Matplotlib 3.3
**Description of work.**
This PR adds support for matplotlib 3.3. It features several changes:

- Fixes some issues with deprecated api e.g (ax.rowNum, ax.colNum)
- missing _set_new_active_manager in figure manager
- QAppThreadCall masking arguments to draw_if_interactive_impl

The main change is restructuring the toolbar class. With Matplotlib 3.3 onwards the toolbar has to be initialised in its constructor, this meant our tool bars inheriting from NavigationToolbarQt were incomptabile. To solve this issue, a new toolbar class has been constructed, which is based on [... cut]
closed 2021-07-12 by martyngigg