mantidproject/mantid #30155: Plot Settings Error Every Resets On GUI To 1 When Applied

Matt-Cumber, 2020-12-14:
The value for error every is correctly saved in the back end it is just not correctly displayed.

### Expected behavior
The value in the GUI should stay as what the users selected

### Actual behavior
The value on the GUI is reset to 1 even if the correct value is applied to the plot

### Steps to reproduce the behavior
1. Load some data
2. Plot a spectrum
3. Open plot settings
4. In Curves, unhide error bars and update error every to something different to 1, then click apply
5. Notice the plot updates but the value on the GUI is 1 again
6. If you try click apply again without doing anything nothing happens, suggesting the value is not actually 1 even if 1 is shown

### Platforms affected
all presumably (found on Mac)

DanielMurphy22, 2021-01-11:
I cannot reproduce this on Windows, but I can on Mac so setting it to Mac only. This is very odd.

DanielMurphy22, 2021-01-11:
This is not broken in 5.1.1 so we'll see if we can fix for this release!

DanielMurphy22, 2021-01-11:
To whoever gets assigned to fix this, look at the error report from Malcolm on 23/12/2020 and decide if this is linked or not. He may have gotten Mantid into an odd state, but the property `errorevery` features front an centre in the stacktrace.

Matt-Cumber, 2021-01-19:
has been fixed by #30359

PR #30359 (2021-01-19): 30155 fix error every updating plot settings
**Description of work.**
Changed how the errorevery property is obtained when getting this property from curve

**To test:**
Load some data and make a plot

1. Open the plot settings and go to the curves tab > error bars
2. Uncheck hide and change the error every property
3. Zoom in on the plot to check this has been applied
4. The value you changed it to should also remain in the GUI (previously reset to 1 every time)
5. Close the settings
6. Reopen them and check value is still correct

Fixes #30155 

---

#### Reviewer ####

Please comment on the [... cut]
closed 2021-01-19 by Matt-Cumber