mantidproject/mantid #30209: Crash In Muon Analysis When You Clear All Workspaces In Mantid

Matt-Cumber, 2020-12-18:
Looks to be trying to retrieve workspaces from the ADS which no longer exist so is crashing. Need some sort of exception to handle if it can't be found instead.

### Expected behavior
The interface should handle this gracefully rather than crash

### Actual behavior
Crashes

### Steps to reproduce the behavior
1. Open Muon Analysis
2. Load in a run (MUSR62260)
3. With the interface remained open, on main workbench window click the clear button in the workspaces toolbox (alternatively File > Clear all memory)
4. Now change to fitting tab on Muon Analysis Interface
5. Crashes

### Platforms affected
all presumably

AnthonyLim23, 2021-01-06:
@Matt-Cumber does this bug still exist? If it does could you fix it

Matt-Cumber, 2021-01-06:
> @Matt-Cumber does this bug still exist? If it does could you fix it

Yes it does. I will take a look

AnthonyLim23, 2021-01-07:
I think we can just close this. I have never seen the clear all used.

closed 2021-01-07 by AnthonyLim23