mantidproject/mantid #30214: Rebin functionality in Elemental Analysis GUI

sf1919, 2020-12-18:
The Grouping Tab in the Elemental Analysis GUI (#29382) requires additional functionality to create Rebin workspaces using information provided through the rebin and rebin_options column of the table. This follows on from PR #30199 .

In addition to this the user dialog input boxes for undertaking a rebin need to be reworded to make it clear the rebin will create a new, separate workspace and not affect the raw data.

The Grouping MVP could further be tidied up by replacing the Observers with GenericObservers.

sf1919, 2021-04-15:
@tolu28-coder - were you able to tackle these issues when you took over work on the Initial PR for the Elemental Analysis Grouping Tab (#30199)?

tolu28-coder, 2021-04-15:
Yes i was able to add the rebin functionality and add the generic observers

sf1919, 2021-04-15:
Did you also reword the dialog input boxes to avoid confusion about how it works?

sf1919, 2021-04-15:
If you could do that so we can close this issues that would be fantastic. I'll assign this one to you in the meantime.

PR #31181 (2021-04-26): Added more tests for new elemental analysis gui
**Description of work.**
This pull request adds extra tests for the new elemental analysis gui, updates the rebin input dialog to inform users that rebininig will create a new workspace and updating grouping tab so as to produce a warning popup when no parameters are given.

**Report to:** Adrian Hillier

**To test:**
1. Before starting Mantid go to https://github.com/mantidproject/mantid/blob/master/Framework/Properties/Mantid.properties.template#L24 , add Muon/Elemental_Analysis_2.py to the end of the line and build Mantid.

2. Download the tes [... cut]
closed 2021-04-26 by DavidFair