mantidproject/mantid #30256: Fit algorithm for multiple domains returns a summed chi squared valued

StephenSmith25, 2021-01-05:
### Expected behavior
Each fit for a multi-domain fit should have its own chi squared value, describing the goodness-of-fit of the function for that dataset.
### Actual behavior
It returns some summed quantity, which doesn't represent the goodness-of-fit. This is more troublesome as each domain could have a different function, so this summed quantity is not representing anything meaningful.
### Steps to reproduce the behavior
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

from mantid.simpleapi import *
import numpy as np

# Create workspaces
x_values = np.linspace(start=1.0,stop=10.0,num=22)
y_values1 = [2.0*x + 10 for x in x_values]
y_values2 = [5.0*x*x + 10.0 for x in x_values]

input_workspace1 = CreateWorkspace(x_values, y_values1)
input_workspace2 = CreateWorkspace(x_values, y_values2)

# Create MultiDomainFunction where datasets have different fitting functions
multi_domain_function = FunctionFactory.createInitializedMultiDomainFunction('name=CompositeFunction', 2)

flat_background = FunctionFactory.createInitialized("name=FlatBackground")
linear_background = FunctionFactory.createInitialized("name=LinearBackground")
exp_decay = FunctionFactory.createInitialized("name=ExpDecay")

composite1 = multi_domain_function.getFunction(0)
composite1.add(flat_background)
composite1.add(linear_background)

composite2 = multi_domain_function.getFunction(1)
composite2.add(flat_background)
composite2.add(exp_decay)

function_string = str(multi_domain_function)

# Perform the fit
fit_output = Fit(Function=function_string,
                 InputWorkspace=input_workspace1, WorkspaceIndex=0, StartX = 1.0, EndX=10.0,
                 InputWorkspace_1=input_workspace2, WorkspaceIndex_1=0, StartX_1 = 1.0, EndX_1=10.0,
                 Output='fit')
print("chi squared", fit_output.OutputChi2overDoF)

```
For this data, the first fit will returned will be a good fit to the data, however the second fit will be poor. This results in a large chi squared value, which doesn't provide insight of the "goodness-of-fit".
### Platforms affected
All

AnthonyLim23, 2021-01-06:
If we have shared parameters between domains (globals), then how would this be reflected in the chi-squared?

StephenSmith25, 2021-01-06:
At the moment, the chi squared for each domain would be summed, but then divided by the total degrees of freedom (so kind of averaged) which I think would only make sense if all of the parameters were shared i.e. the same model is used for all datasets. The way I see it (maybe wrong here), if we have a model M_k for each dataset D_k, then our hypotheses are:

H_0 : The model M_k fits the data D_k
H_A : The model doesn't fit (or some other model does)

Then we use the chi_squared test, compared to the chi_squared distribution, to make a decision on the hypothesis. I feel like if we arbitrarily sum them, we are not testing the suitability of the model M_k, but the average suitability of all of the models over the domain, which would only make sense if each model was the same. I think it would be better to report k chi_squared values each referring to the kth model and dataset. 

 Currently, in the example in the script, one of the models would pass the chi_squared test, but the other would fail, but the resulting chi squared value would signify that the overall fit has failed.

AnthonyLim23, 2021-01-06:
I understand, but equally feel that it could be misleading when some parameters are shared. When sharing parameters you are effectively fitting to more data, which would not be included in the chi-squared for a single data set. The simple example of a shared flat background and some other function(s). Then the chi-squared could be worse for data set 1, as the flat background was dragged up by data set 2. 

I think we should talk to Tyrone's group about this. They specialise in stats and fitting.

StephenSmith25, 2021-01-06:
I agree, but maybe the statistic we report should not be chi squared then, as chi squared is intrinsic to the least squares fit, for instance, to find errors / confidence on parameters we find the change in parameter which would cause chi squared to increase from its minimum by a certain amount. This is found by doing a Taylor series expansion of the chi_squared function about the minimum chi squared, which produces an n-dimensional ellipse in the parameter space:

[image]

Its common to look at the errors which would cause chi squared to increase by chi_squared, e.g chi squared min=1.43 and we find parameter errors that increase chi squared by 1.43. If we use some averaged chi squared across the whole fit would we get the correct error bars? 

This probably just needs more thought, so I agree, discussing this with another group would be worthwhile

AnthonyLim23, 2021-01-06:
I have sent Tyrone a message, will let you know when he gets back to me. Then we can set up a meeting.

If this is going to be a major change, would it be worth considering a new fitting algorithm? Mantid's is getting old and has some issues.

StephenSmith25, 2021-01-06:
Great. And yes, I would say a new fit algorithm would be ideal - but this would need a discussion and plan going forward - At the moment the fit algorithm is very unmanagable (circa 22 inputs and 10 outputs).

AnthonyLim23, 2021-01-06:
Set something up with me and Martyn to discuss the possibilities.

closed 2021-07-12 by stale[bot]