mantidproject/mantid #30268: Remove Qt4-only code

martyngigg, 2021-01-07:
MantidPlot was removed from distribution during the 6.0 development cycle but the code and its dependencies still remain. This is an umbrella issue for removing the Qt4-only code from the codebase.

Known tasks:
* Remove MantidPlot
* Remove Qt4 CMake targets & tests in C++/Python
  * Remove any code, particular in common widgets that is not used in Qt5
* Remove ifdefs in Qt5 code that supports Qt4
* Remove `ENABLE_MANTIDPLOT` CMake option

peterfpeterson, 2022-03-10:
This should not be closed

martyngigg, 2023-01-24:
The final steps are to review if anything is remaining and either finish it or just close this issue.

robertapplin, 2023-02-02:
I think this can be closed now that #35082 is merged

closed 2023-02-02 by robertapplin