mantidproject/mantid #30286: Hang when overwriting MD workspace with sliceviewer open

gemmaguest, 2021-01-07:
### Expected behavior
You should be able to open sliceviewer on a workspace and replace the workspace in the ADS

### Actual behavior
Workbench hangs if this is done on an MD workspace

### Steps to reproduce the behavior
- Run the script below
- Show sliceviewer on QxQy_1
- Run the script again. Workbench hangs.

Note that it seems ok with a workspace that is not an MD workspace (if I uncomment the OutputAsMDWorkspace=False property then it's ok).

```
name = "POLREF4699"
PIX=1.1E-3 #m
SC=75
avgDB=29
Load(Filename=name,OutputWorkspace=name)
ConvertUnits(InputWorkspace=name,OutputWorkspace=name,Target="Wavelength",AlignBins="1")
CropWorkspace(InputWorkspace=name,OutputWorkspace='Io',XMin=0.8,XMax=14.5,StartWorkspaceIndex=2,EndWorkspaceIndex=2)
CropWorkspace(InputWorkspace=name,OutputWorkspace='D',XMin=0.8,XMax=14.5,StartWorkspaceIndex=3)
Divide(LHSWorkspace='D',RHSWorkspace='Io',OutputWorkspace='I', AllowDifferentNumberSpectra='1',ClearRHSWorkspace='1')
MoveInstrumentComponent(Workspace='I',ComponentName="lineardetector",X=0,Y=0,Z=-PIX*( (SC-avgDB)/2.0 +avgDB) )
ConvertSpectrumAxis(InputWorkspace='I',OutputWorkspace='SignedTheta_vs_Wavelength',Target='signed_theta')
QxQy, _QxQy_vertexes = ConvertToReflectometryQ(InputWorkspace='SignedTheta_vs_Wavelength',
                                               #OutputAsMDWorkspace=False,
                                               OutputDimensions='Q (lab frame)', Extents='-0.0005,0.0005,0,0.12')
```

### Platforms affected
Seen on Ubuntu 18.04 in master. It's ok in 5.1.1

rosswhitfield, 2021-01-07:
Is this fixed by #30261 ?

gemmaguest, 2021-01-08:
> 
> 
> Is this fixed by #30261 ?

@rosswhitfield Yes, perfect, thanks!

closed 2021-01-08 by gemmaguest