mantidproject/mantid #30289: Add back to back exponential parameters to prm file for calibrated banks in EngDiff UI

RichardWaiteSTFC, 2021-01-08:
Add back to back exponential parameters to. prm file (*input file for GSAS) for calibrated banks in EngDiff UI

DanielMurphy22, 2021-02-17:
Here's a summary of how far I've gotten.

In the calibration model we [create an output prm file](https://github.com/mantidproject/mantid/blob/master/scripts/Engineering/gui/engineering_diffraction/tabs/calibration/model.py#L271) by copying the [template file](https://github.com/mantidproject/mantid/blob/master/scripts/Engineering/template_ENGINX_241391_236516_North_and_South_banks.prm), and [simply replacing the DIFA, DIFC and TZERO values](https://github.com/mantidproject/mantid/blob/master/scripts/Engineering/EnggUtils.py#L354). This means that for all output prm files the alpha, beta, sigma coefficients are simply copied from the template.

For the ENGINX [template file](https://github.com/mantidproject/mantid/blob/master/scripts/Engineering/template_ENGINX_241391_236516_North_and_South_banks.prm) the values are (starting from line `INS  1PRCF11`). Alternatively, the (presumably correct and up-to date) values in the [ENGINX parameters.xml](https://github.com/mantidproject/mantid/blob/master/instrument/ENGIN-X_Parameters.xml#L4-L14).

| Values from| ALPHA |           BETA-0        |         BETA-1   |     SIGMA0-SQRD  |    SIGMA1-SQRD   |     SIGMA2-SQRD |
|-----|---|----|-----|------|------|----------|
| [Template](https://github.com/mantidproject/mantid/blob/master/scripts/Engineering/template_ENGINX_241391_236516_North_and_South_banks.prm) | 0.274872E+00 |  0.179289E-01 |  0.379667E-01 |  0.388241E+03   |   0.768656E+02    |   0.432991E+02 |
| [Parameters](https://github.com/mantidproject/mantid/blob/master/instrument/ENGIN-X_Parameters.xml#L4-L14) | 0.968000E-01 |  0.216000E-01 |  0.123000E-01 |  0.000000E+00   |   0.821321E+02    |   0.425988E+02 |

@RichardWaiteSTFC is the correct solution to write over the template, as we do for DIFA, DIFC and TZERO, with the coefficients from the parameters.xml file? If yes, should it only be read in for data after the valid-from date in parameters.xml? Also, am I right that SIGMA0 is 0 in the parameters file?

Just a note for myself, the equations half-way down this page correspond with the parameters file and the equations in the GSAS technical manual pg155

RichardWaiteSTFC, 2021-02-17:
@DanielMurphy22 thanks for the nice summary. Yes we should aim to overwrite the values in the template with the ones in the Parameters.xml (which we will keep up to date). SIGMA0 = 0 is deliberate - in my initial refinement it was zero within error and I believe it is not uncommon for it to be fixed at 0 (e.g. it is also 0 in `HRPD_Parameters.xml`).
I am a bit concerned about the disparity between the .prm and the Parameters.xml. I would expect that the SIGMAS might be different given I fixed sigma0 = 0 (though the magnitude of SIGMA0 is larger than I was expecting) but the alpha and beta coefficients are also quite different from the ones in Parameters.xml (which agree well with the ceria data in run 193749). Given we don't know where the parameters in the .prm template came from (perhaps not a standard sample) it is probably worth checking if you put the parameters from the .prm into the Parameters.xml the estimated BackToBackExponential parameters still give a decent guess.

RichardWaiteSTFC, 2021-02-17:
Here is some focussed ceria data post calibration of the instrument [ENGINX193749_NorthBank_postInitialCalibration.zip](https://github.com/mantidproject/mantid/files/5996946/ENGINX193749_NorthBank_postInitialCalibration.zip). I have edited the instrument geometry so that when you load in the files it will complain there is no such instrument - don't worry I have called it NorthBank which corresponds to the component in the ENGINX_Parameters.xml. This means LoadParameterFile will work, but it won't know it's ENGIN-X at the point it's loaded so it will not have any parameter info automatically associated with it. For what you need to do this is actually a good thing as you can be certain which Parameter.xml you load onto it.

closed 2021-03-01 by martyngigg