mantidproject/mantid #30296: SaveNXSPE can save files that raise error on LoadNXSPE

ajjackson, 2021-01-08:
### Expected behavior

SaveNXSPE and LoadNXSPE are symmetric, and a workspace saved to file with the SaveNXSPE algorithm will be more or less identical to a workspace loaded from this file with LoadNXSPE.

If a workspace does not contain the information required for an acceptable NXSPE file, a warning/error should occur before writing the file.

### Actual behavior

If the Workspace is created for a new model instrument following an example in the [EditInstrumentGeometry docs](https://docs.mantidproject.org/nightly/algorithms/EditInstrumentGeometry-v1.html), the file is saved succesfully. When loaded, an Error is raised complaining that no instrument with this name could be found in any facility.

### Steps to reproduce the behavior

```python
ws = CreateWorkspace(
    DataX=[0., 1., 2., 3.],
    DataY=np.arange(9).reshape(3, 3),
    UnitX='DeltaE',
    NSpec=3)
EditInstrumentGeometry(
    ws,
    PrimaryFlightPath=5.,
    SpectrumIDs=[1, 2, 3],
    L2=[2.0, 2.3, 2.6],
    Polar=[10.0, 15.0, 23.0],
    Azimuthal=[0.0, 0.0, 0.0],
    DetectorIDs=[100, 101, 102],
    InstrumentName='Bizarrio')
SaveNXSPE(InputWorkspace=ws, Filename='bizarrio_data.nxspe', Efixed=5.)
LoadNXSPE('bizarrio_data.nxspe', OutputWorkspace='imported-ws')
```

which for me yields the warning
```
NaN encountered while trying to push axis to goniometer, Operation aborted
axis namepsi
axisx0
axisy0
axisz0
```
and the error
```
Error in execution of algorithm LoadNXSPE:
Failed to find an instrument with this name in any facility: 'Bizarrio' - search object Bizarrio
RuntimeError: Failed to find an instrument with this name in any facility: 'Bizarrio' - search object Bizarrio
  File "/mantid_data/scratch/nxspe_roundtrip.py", line 21, in <module>
    LoadNXSPE('bizarrio_data.nxspe')
  File "/mantid_src/Framework/PythonInterface/mantid/simpleapi.py", line 1036, in __call__
    algm.execute()
```

### Platforms affected
Encountered on Ubuntu Docker build, but no reason to suspect this is platform-dependent.

ajjackson, 2021-07-16:
The issue requires input from Core and/or MSLICE. But it is still a genuine issue, related to the tight coupling between workspaces and instrument data. It should be easier to use workspaces that do not represent existing instruments with a large amount of metadata.

SilkeSchomann, 2022-11-28:
@ajjackson I've opened a branch with a fix for psi error. There is an entry for psi in the NXSPE file that is read as nan and subsequently causes an error when configuring the Goniometer.

Fixing the runtime error would require a design change in ConfigService. Alternatively, there might be an easy workaround. The error is thrown when ConfigService fails to find this instrument for any known facility in order to provide the corresponding instrument parameter file name. LoadNXSPE only searches for the instrument parameter file when the instrument name is neither empty nor set to "NXSPE".

ajjackson, 2022-11-28:
That seems to help a lot! I note that the DetectorIDs of the re-imported data are 1, 2, 3 rather than 100, 101, 102 but maybe that makes sense within some other Mantid convention?
[image]

This does seem to allow me to import custom data into MSlice which is just what was needed. I suggest then that we tweak the SaveNXSPE algorithm to error/warn if an unknown instrument name is set, and (if only warning) set the parameter to "NXSPE" before writing.

SilkeSchomann, 2022-11-28:
> That seems to help a lot! I note that the DetectorIDs of the re-imported data are 1, 2, 3 rather than 100, 101, 102 but maybe that makes sense within some other Mantid convention?

SaveNXSPE is designed to not store the DetectorIDs. LoadNXSPE just assigns ascending numbers starting with 1.

https://docs.mantidproject.org/nightly/algorithms/SaveNXSPE-v1.html

SilkeSchomann, 2022-11-28:
> This does seem to allow me to import custom data into MSlice which is just what was needed. I suggest then that we tweak the SaveNXSPE algorithm to error/warn if an unknown instrument name is set, and (if only warning) set the parameter to "NXSPE" before writing.

That would be one option. Currently LoadNXSPE sets the instrument name to "NXSPE" when the instrument name property is empty.

ajjackson, 2022-11-29:
I don't see any purpose for allowing an unreadable NXSPE to be written, it seems most helpful to fail immediately with a useful error message.

I suppose the main argument to go with "raise warning, change name" is that not everybody reads error messages so in practice this gets in the way of less science 😞.

closed 2023-01-12 by SilkeSchomann