mantidproject/mantid #30299: SaveNexus/LoadNexus do not work with custom detectors from EditInstrumentGeometry

ajjackson, 2021-01-08:
### Expected behavior

A workspace with custom instrument/detector properties can be saved to Nexus with SaveNexus. When a workspace is obtained from this file with LoadNexus, methods and algorithms accessing the detector information should give the same results for either workspace.

My use case is that I would like to generate workspaces from Abins that correspond to simplified instruments with fewer detectors than real ones, and use them with inbuilt Mantid algorithms such as ConvertUnits and SofQWNormalisedPolygon. I am running into all kinds of trouble where these algorithms require instrument/sample data to exist, and cannot simply load an existing instrument because the angle ranges and numbers of spectra do not match. The documentation of EditInstrumentGeometry shows that a fictitious instrument can be set up and attached to a workspace - but in practice the resulting workspace does not seem to be fully functional.

### Actual behavior
When a workspace produced with customised Instrument Geometry (following an example in the [EditInstrumentGeometry docs](https://docs.mantidproject.org/nightly/algorithms/EditInstrumentGeometry-v1.html)) is saved to disk, a warning is raised

> Saving Instrument with no XML data. If this was instrument data you may not be able to load this data back into Mantid, for fitted/analysed data this warning can be ignored.

and when the file is reloaded, another warning is raised

> Error loading Instrument section of nxs file
> Failed to find an instrument with this name in any facility: 'Bizarrio' - search object Bizarrio
> Try running LoadInstrument Algorithm on the Workspace to update the geometry

I cannot find any procedure to generate/write the XML data from the edited/novel instrument geometry. 
I believe I need EditInstrumentGeometry to also create/modify XML data associated with the Workspace, so that this is available on save/load.

### Steps to reproduce the behavior

```python
ws = CreateWorkspace(
    DataX=[0., 1., 2., 3.],
    DataY=np.arange(9).reshape(3, 3),
    UnitX='DeltaE',
    NSpec=3)
EditInstrumentGeometry(
    ws,
    PrimaryFlightPath=5.,
    SpectrumIDs=[1, 2, 3],
    L2=[2.0, 2.3, 2.6],
    Polar=[10.0, 15.0, 23.0],
    Azimuthal=[0.0, 0.0, 0.0],
    DetectorIDs=[100, 101, 102],
    InstrumentName='Bizarrio')

# No errors when these are called on the new workspace
CreateDetectorTable(InputWorkspace='ws',
                    DetectorTableWorkspace='ws-Detectors')
print(ws.getDetector(0))

# Raises warning
SaveNexus(ws, 'bizarrio-data.nxs')

# Raises warning
ws_roundtrip = LoadNexus('bizarrio-data.nxs')

# This raises an error now the data was reloaded
CreateDetectorTable(InputWorkspace='ws_roundtrip',
                    DetectorTableWorkspace='ws_roundtrip-Detectors')

# So does this
print(ws_roundtrip.getDetector(0))
```

### Platforms affected
All, presumably?

ajjackson, 2021-01-08:
(Many thanks to @mducle, @SilkeSchomann @RichardWaiteSTFC @martyngigg for getting me to this point... I suspect this is not the only remaining obstacle to using SofQWNormalisedPolygon, but it seems like a useful point along the way to raise an Issue. And I _would_ also like Abins users to be able to save their workspaces.)

ajjackson, 2021-01-11:
I'm also getting the "no XML data" from data generated using CreateSampleWorkspace. (Although the parameter space for this algorithm is pretty big, so maybe there's a way...)

ajjackson, 2021-01-18:
Update from discussion with @martyngigg :
It is expected that SaveNexusESS will have better support for custom instruments as it should write geometry information to the file.

In my test of this, there is no error on save, but problems remain on load:

```python
ws = CreateWorkspace(
    DataX=[0., 1., 2., 3.],
    DataY=np.arange(9).reshape(3, 3),
    UnitX='DeltaE',
    NSpec=3)
EditInstrumentGeometry(
    ws,
    PrimaryFlightPath=5.,
    SpectrumIDs=[1, 2, 3],
    L2=[2.0, 2.3, 2.6],
    Polar=[10.0, 15.0, 23.0],
    Azimuthal=[0.0, 0.0, 0.0],
    DetectorIDs=[100, 101, 102],
    InstrumentName='Bizarrio')
    
SaveNexusESS(ws, 'bizarrio.nxs')

ws_reloaded = LoadNexusProcessed('bizarrio.nxs')
```

which raises
```
Error loading Instrument section of nxs file
NXopengroup(instrument, NXinstrument) failed
Try running LoadInstrument Algorithm on the Workspace to update the geometry
MatrixWorkspace::setIndexInfo: IndexInfo size does not match number of histograms in workspace
```

It is unclear whether the problem lies in SaveNexusESS or LoadNexusProcessed.

SilkeSchomann, 2024-11-18:
Recent changes in SaveNexusESS have fixed this problem (nightly build from 11/11/24).

closed 2024-11-18 by SilkeSchomann