mantidproject/mantid #30303: ISIS SANS: Browse button in sum runs causes exception

StephenSmith25, 2021-01-11:
### Expected behavior
Exiting the browse option without selecting any files should not produce an error.
### Actual behavior
It produces an unhandled exception
### Steps to reproduce the behavior

1. Open the ISIS SANS GUI
2. Go to sum runs
3. Click browse
4. Click cancel
5. You should get an unexpected exception error message

### Platforms affected
All

StephenSmith25, 2021-01-11:
Tested on windows: Found two issues 

#30306
#30303

gemmaguest, 2021-01-11:
This has been reported before but for some reason I thought that was fixed. I can reproduce it again now, though. It's been a problem in workbench for some time. I also get the exception when selecting a file.

PR #30366 (2021-01-18): ISIS SANS Fix crash when browsing files in sum run tab 
Fixes a crash in the ISIS SANS sum run gui when the user is browsing files, and then presses cancel. This was due to the Qt api for `QFileDialog.getOpenFileNames` changing.

**To test:**
1. Open ISIS SANS
2. Go to sum runs and click browse
3. Cancel the file picker dialog, verify that no exceptions occur
4. Click browse and select a file, verify that no exceptions occur (you will have to set an instrument in the first tab for this.)

Fixes #30303 
Related: #27530

---

#### Reviewer ####

Please comment on the following ([full description [... cut]
closed 2021-01-19 by joseph-torsney