mantidproject/mantid #30306: ISIS SANS: Settings reset when table values are modified

StephenSmith25, 2021-01-11:
Following the SANS Gui test instructions (https://developer.mantidproject.org/Testing/SANSGUI/SANSGUITests.html), I've noticed issues in steps 4-5 in the Runs table editing test:

_4. Try changing a random setting in the settings tab and remember what you set it to.
5. Make some more edits to the table. Check that the setting is still showing the value you changed it to, rather than the original._

Currently, if I change the value of a setting and edit the table it switches back to the original setting value.

### Expected behavior
The new value of the setting should be kept between table edits
### Actual behavior
It resets.
### Steps to reproduce the behavior

1. Open the ISIS Sans GUI
2. Assuming you have the training data load the user file /loqdemo/MaskFile.toml and the batch file /loqdemo/batch_mode_reduction.csv
3. Change a setting e.g Q,Wavelength, DetectorLimits/Radius Limit
4. Go back to the runs tab and change one of the output names 
5. Go back to the setting, it will have reset to its original value

### Platforms affected
All

gemmaguest, 2021-01-11:
I think this was me pre-emptively adding a test to the instructions (in PR #29632) for something that has been broken for a long time, which we expected to be fixed by this release, but it ended up not getting fixed. It's not a high priority so we don't need to worry about it for this release. However, I'll add a note to the test instructions to explain the situation so this doesn't take up testers' time again (sorry about that).

gemmaguest, 2021-03-26:
Apparently this was working in v5.0 so must have been introduced in v5.1

gemmaguest, 2021-05-04:
I think I have a fix for this but I am not convinced it is safe to put into this release so pushing back to the next milestone

closed 2025-05-17 by github-actions[bot]