mantidproject/mantid #30317: Reduction fails second time through from GUI

gemmaguest, 2021-01-11:
### Expected behavior
You should be able to repeat the same reduction and get the same results

### Actual behavior
For some runs, running a second time fails

### Steps to reproduce the behavior
- Open the ISIS Reflectometry interface
- Go to the menu Batch->Load and from the ISIS sample data select the file INTER_NR_test.json
- Click Process to process all rows
- In Workbench, click Clear to delete all workspaces
- In the interface, click Process to process all rows again. The rows in the second group fail with `CalculateFlatBackground: Infinte number or NaN found in input data`.

### Platforms affected
Probably all.

This exists in v5.1.1 so is not a regression in the current release, but it is worrying so I am marking for milestone 6.0 for now until I investigate further.

gemmaguest, 2021-01-13:
This looks like a bug in `WorkspaceSelector`. The reflectometry GUI relies on the flood workspace selection being optional, i.e. containing a [blank entry](https://github.com/mantidproject/mantid/blob/master/qt/widgets/common/src/WorkspaceSelector.cpp#L309). However, when you clear the ADS, this is [removed](https://github.com/mantidproject/mantid/blob/master/qt/widgets/common/src/WorkspaceSelector.cpp#L203). The next time a workspace is created (e.g. by processing the first row) it gets populated as the flood workspace, causing subsequent reductions to fail.

@martyngigg I can easily fix this by adding `if (m_optional) addItem("")` to `WorkspaceSelector::handleClearEvent` but I'm not sure it's safe to change `WorkspaceSelector` this late in the release?

martyngigg, 2021-01-13:
Ah yes I see. It does feel risky to modify the workspace selector at this stage.
Could we create two fixes?
 1. the workaround as you suggest in your own observer for the release
 2. the "proper" fix in the `WorkspaceSelector` that can go to `master` once the release branch exists?

PR #30354 (2021-03-04): Fix WorkspaceSelector so it does not remove optional blank row
This PR fixes a bug where the blank row in the workspace selector (which is required to make the selection of a workspace optional) is removed when the ADS is cleared.

This PR also removes the workaround from PR #30353 for a problem on the reflectometry GUI that was caused by this. 

Fixes #30317

**To test:**

Run the test in #30353

*This does not require release notes* because **the symptoms were already fixed and documented by the workaround**

---

#### Reviewer ####

Please comment on the following ([full description](http:/ [... cut]
closed 2021-03-04 by martyngigg