mantidproject/mantid #30318: Update ISIS SANS GUI manual test instructions with note on bug

gemmaguest, 2021-01-11:
The unscripted testing instructions for ISIS SANS should be updated with a note to explain that there is a current bug (the documented behaviour is what it should be so I don't want to remove it, but we should update the doc to explain the situation to save testers' time in future). See the issue and comment on #30306 for details.

DanielMurphy22, 2021-04-13:
I've not edited the script for producing the Manual tests, but I've added a comment to https://github.com/mantidproject/mantid/issues/31120

gemmaguest, 2021-05-04:
I'm going to close this because unscripted testing has already been done for this release and the bug should be fixed by the next release

closed 2021-05-04 by gemmaguest