mantidproject/mantid #30325: Revise Indirect Data Analysis documents

Harrietbrown, 2021-01-12:
We have been made aware that there are users doin QENS that are not aware that they can make use of our indirect tools. this issue is to revise the documentation of the interface to try and make it more clear what it is capable of.

Report to @SpencerHowells

AnthonyLim23, 2021-01-12:
I think we should just add a few docs to the direct part that link to the indirect docs

Harrietbrown, 2021-03-02:
as part of this I should look into #22837

AnthonyLim23, 2021-08-20:
@Harrietbrown and @SilkeSchomann have we done this or do we have more left to do?

AnthonyLim23, 2022-01-05:
if we have some time can we work on these for the release?

Harrietbrown, 2023-01-04:
Elements to focus on for this issue are:

Create interface structure doc for Inelastic data Manipulation like what we have for Indirect Data Analysis https://github.com/mantidproject/mantid/blob/main/dev-docs/source/IndirectDataAnalysisFileStructure.rst

How to add a fit function in data analysis tabs

How to use the workspace data table widget.

PR #35071 (2023-01-26): Update Indirect dev documentation
**Description of work.**
This PR adds documentation for the indirect data analysis and indirect data manipulation of windows class structure. Also, it adds documentation for how to add a fit function to different tabs.

**To test:**
- build `dev-docs-html`

Fixes #30325. 

*This does not require release notes* because **it is a dev doc update**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the [... cut]
closed 2023-01-26 by martyngigg