mantidproject/mantid #30333: Default load path on Fitting tab not correct if run number used during Focus

DannyHindson, 2021-01-12:
The default path in the "Focused Run Files" on the fitting tab is missing a "User" level if the last focus had an RB number specified. This gave a "file not found" message when hitting Load

On Windows for example, the Focus creates files in this folder if a run number is used:

C:\Users\iao46725\Engineering_Mantid\User\<run number>\Focus\

martyngigg, 2021-01-19:
Can I just check if we would still like to get this into 6.0?

DannyHindson, 2021-01-19:
> Can I just check if we would still like to get this into 6.0?

I'd like to get the fix done in this release sprint and into the nightly build but I don't think its essential that it goes into the release. @RichardWaiteSTFC, @clayton-tom - do shout if you disagree

PR #30438 (2021-01-28): Fix bug to ensure engdiffui fit tab gets correct focus filename
**Description of work.**
The filename input in the fitting tab is supposed to update with the most recently focused .nxs file, but this would only work in the event that an rb number was not supplied, as this would change the directory. This fixes that to ensure that the correct filename will be displayed in any case

**To test:**
1. Interfaces -> Diffraction -> Engineering Diffraction
2. Load a calibration file e.g
[ENGINX_307521_305738_all_banks.zip](https://github.com/mantidproject/mantid/files/5848295/ENGINX_307521_305738_all_ [... cut]
closed 2021-01-28 by gvardany