mantidproject/mantid #30360: sliceviewer crash displayingUnits for Direct data

DanielMurphy22, 2021-01-14:
**Original reporter:** Tatiana and @mducle 

### Actual behavior
Hard crash when trying to calculate the cursor data for sliceviewer for .nxspe data.
### Steps to reproduce the behavior
Load `MAR27689_Ei19.95meV.nxspe`
Open sliceviewer
Possibly, move mouse over data image or try to zoom in.
### Platforms affected
Presumably all, found on Linux

mducle, 2021-01-14:
The crash happens in [`ImageInfoModelMatrixWS.cpp`](https://github.com/mantidproject/mantid/blob/master/qt/widgets/common/src/ImageInfoModelMatrixWS.cpp#L140) which tries to update the value of `efixed` to field which doesn't exist. The code mentions `// the final parameter is unused and a relic` but it seems a recent code change has eliminated this field and this now causes an indexing error (and segfault).

Stack trace:

```
(gdb) bt
#0  0x00007effe5f54ff7 in QString::operator=(QString const&) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007effd2cd062a in MantidQt::MantidWidgets::ImageInfoModelMatrixWS::setUnitsInfo(MantidQt::MantidWidgets::ImageInfoModel::ImageInfo*, int, unsigned long, double) const ()
    at /opt/Mantid/lib/mantidqt/../libMantidQtWidgetsCommonQt5.so
#2  0x00007effd2cd1134 in MantidQt::MantidWidgets::ImageInfoModelMatrixWS::info(double, double, double) const () at /opt/Mantid/lib/mantidqt/../libMantidQtWidgetsCommonQt5.so
#3  0x00007effd2cd5276 in MantidQt::MantidWidgets::ImageInfoPresenter::cursorAt(double, double, double) () at /opt/Mantid/lib/mantidqt/../libMantidQtWidgetsCommonQt5.so
#4  0x00007effd3215ae9 in meth_ImageInfoWidget_cursorAt () at /opt/Mantid/lib/mantidqt/_commonqt5.so
#5  0x000000000050a12f in  ()
#6  0x000000000050beb4 in _PyEval_EvalFrameDefault ()
```

The fix is to add a new field to `displayUnits` (and also to get rid of the `requiresEFixed` flag and refactor `setUnitsInfo` to not use it.)

SpencerHowells, 2021-01-14:
@robertapplin The Direct reduction programs did not set efixed. In particular ChangeUnits did not work. Indirect programs did not like this either- as a result this has now been corrected. However old files will still have this problem. The cure was to use SetInstrumentParameter to set efixed to the value of Ei in the SampleLogs. This could be used here as well.

mducle, 2021-01-14:
I don't think it's an issue of getting the value of `efixed` - I get the same crash even after applying `SetInstrumentParameter(data_ws,ParameterName='EFixed',ParameterType='Number', Value='19.95')` on the workspace.

Rather I think it's because the code assumes that there are `n+1` fields in the information strip at the top of the SliceViewer, where `n` is the number of fields defined by `createItemNames` and `updateUnits`. The comment in the code suggested that previously this was true but it seems to be no longer true.

This `n` seems to be hard-coded into the code - so I suggest just changing the code to have an additional field... I had thought that the number of fields is dependent on the type of workspaces plotted but it seems to be the same for all workspaces.

robertapplin, 2021-01-14:
@mducle please could you upload the `.nxspe` data? I don't have any

robertapplin, 2021-01-14:
Nevermind, I found some!

robertapplin, 2021-01-15:
Has been fixed

PR #30370 (2021-01-15): Fix a SliceViewer crash when x axis is not labelled TOF
**Description of work.**
This PR fixes a crash on the SliceViewer when hovering the cursor over the loaded direct/indirect data. This occurs when the x unit of the loaded workspace isn't TOF.

The PR adds a test that would have previously failed.

Thanks to @mducle and @DanielMurphy22 for their help.

**To test:**
1. Load `MAR27689_Ei19.95meV.nxspe`
2. Show data in Sliceviewer
3. Move mouse over data image or try to zoom in.
4. There should be no crash!

**Data**
[MAR25087_Ei40.00meV.zip](https://github.com/mantidproject/mantid/files/5815 [... cut]
closed 2021-01-15 by robertapplin