mantidproject/mantid #30402: ErrorReporter: handle large text inputs better

DanielMurphy22, 2021-01-18:
For the errorreporter, there are problems when the character limit for a textBox (3200) or stacktrace (10000) is exceeded. This has been found by recent linode app errors (see slack channel).

### textBox
Currently trying to send a report with text > 3200 characters fails with status 500:
[image]

This is okay, but it would be better to tell the user that the word limit for the textBox has been exceeded. 

### stacktrace
However, for a stacktrace > 10,000 characters, it is probably worth taking the start and end (most useful bits) of the stacktrace and leaving the middle.

ConorMFinn, 2021-12-16:
This could do with being resolved. Copying, say, a stack trace from the MacOS error reporter can easily exceeed the 3200 character limit and the error reporter reports the server error and closes without letting the user amend their message. This isn't great UX and could result in useful crash information never making it to us.

DanielMurphy22, 2022-11-29:
Reopening as just received the same error in the linode-app-errors Slack channel: `django.db.utils.DataError: value too long for type character varying(3200)`

jhaigh0, 2022-12-15:
Is there a good way to create a long error message to test this?

ConorMFinn, 2022-12-15:
If you mean for writing unit tests, you can just do `test_string = "x" * 10001`. If you need it outside, you could run a little python script to add that to the clipboard:
```
import subprocess 
subprocess.run("clip", text=True, input="x"*10)
```
I think it's `clip` for windows, it's `pbcopy` on mac

DanielMurphy22, 2022-12-16:
If you're looking to live test MantidWorkbench, I think the following is possible: 
Pretty sure when the errorreporter appears for the user we have created a stacktrace.txt file in appdata. This is read into the Show details dialog on the errorreporter. When the errorreport is sent, this locl file is renamed stacktrace_sent.txt
https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/dialogs/errorreports/presenter.py

closed 2023-01-12 by thomashampson