mantidproject/mantid #30406: Unhandled exception closing sliceviewer on live data

gemmaguest, 2021-01-19:
### Expected behavior
You should be able to open and close sliceviewer on live data without problems

### Actual behavior
Sometimes there is an unhandled exception closing sliceviewer when live data is being rapidly updated

### Steps to reproduce the behavior
- In the Settings, set the default instrument to TEST_LIVE
- Run the algorithm FakeISISHistoDAE
- Go to Load->Live data and start live data monitoring with instrument ISIS_Histogram, a rapid update interval e.g. `0.1`, accumulation method Replace and no processing or post-processing.
- Open and close slice viewer on the output workspace. You may need to repeat a few times but quite soon I get the following exception:

```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\sliceviewer\presenter.py", line 381, in refresh_view
    self.view.data_view.image_info_widget.setWorkspace(self.model._get_ws())
RuntimeError: wrapped C/C++ object of type ImageInfoWidget has been deleted
```

### Platforms affected
Seen on Windows in the current release candidate. I can't reproduce it in 5.1.1. I suspect it is related to the change that makes sliceviewer update when the underlying data changes.

gemmaguest, 2021-01-19:
I think there's a race condition that goes like this:
1. The ADS replace handler calls [delayed_refresh](https://github.com/mantidproject/mantid/blob/master/qt/python/mantidqt/widgets/sliceviewer/view.py#L581).
2. You close sliceviewer before the timer expires. The first thing it does is [clear the ADS observers](https://github.com/mantidproject/mantid/blob/master/qt/python/mantidqt/widgets/observers/observing_view.py#L47), but the call to `presenter.refresh_view` is already queued.
3. `presenter.refresh_view` tries to use the view, which is now deleted

I think #30261 has probably exacerbated this, but I can still reproduce the problem with the 0ms delay. If I remove the timer it seems to work, but I'm sure this was added for good reason.

martyngigg, 2021-01-20:
Ah yes I see. [This commit](https://github.com/mantidproject/mantid/commit/0954a4a0ff9d3b3616f904c801a98899703b3779) introduced the QTimer and attempts to explain the reason. Reproduced here for easier reading:

```
Updating the model when an MDEventWorkspace is changed
requires running BinMD. The ADS observer
is fired as part of the algorithm execution and for
replacement the algotithm has not yet released the
workspace lock so trying to run another algorithm
throws an exception. We avoid this by delaying the
view update until the next event loop cycle and the
observer has then completed.
```

so we are essentially trying to run an algorithm inside another one before it has completed.

gemmaguest, 2021-02-01:
Fixed by #30457

PR #30457 (2021-01-28): Fix sliceviewer exception with ADS update when closing
**Description of work.**

This PR fixes an unhandled exception that can occur when the sliceviewer is refreshed due to an ADS update but the user closes the sliceviewer window before the refresh is complete. The fix for now is simply to check that the view exists before refreshing it. Longer term it may be better to see if we can remove the race altogther.

**To test:**
Test on Windows.
- In the Settings, set the default instrument to TEST_LIVE
- Run the algorithm FakeISISHistoDAE
- Go to Load->Live data and start live data monitoring with [... cut]
closed 2021-02-01 by gemmaguest