mantidproject/mantid #30423: Make sure all Qt4 tests are also ran in Qt5

Harrietbrown, 2021-01-20:
With the removal of mantidplot all Qt4 tests are not being ran, we need to make sure that all previous tests that we still need to be ran are being run in Qt5 as well

Harrietbrown, 2021-01-21:
`MantidQtWidgetsPlottingTest` and `MantidQtWidgetsSliceViewerTest` are not in Qt5 at the moment.

Harrietbrown, 2021-01-21:
It has also been noted that in `MantidQtScientificInterfacesTestQt5` the cmake list has a different list of test files for qt4 and qt5

peterfpeterson, 2021-01-22:
From [scripts/test/CMakeLists.txt](https://github.com/mantidproject/mantid/blob/master/scripts/test/CMakeLists.txt#L30-L37):
* SANSBatchModeTest.py
* SANSCentreFinderTest.py
* SANSCommandInterfaceTest.py
* SansIsisGuiSettings.py
* SANSReducerTest.py
* SANSReductionStepsUserFileTest.py
* pythonTSVTest.py

Matt-Cumber, 2021-01-25:
ALC tests will be covered by #30466

martyngigg, 2021-01-27:
> `MantidQtWidgetsPlottingTest` and `MantidQtWidgetsSliceViewerTest` are not in Qt5 at the moment.

Neither of these should need porting as they'll be dropping in favour of the Qt5 alternatives.

martyngigg, 2021-01-27:
> It has also been noted that in `MantidQtScientificInterfacesTestQt5` the cmake list has a different list of test files for qt4 and qt5

Do we know if there are more missing than the ALC ones?

martyngigg, 2021-01-27:
I'm going to remove the "High Priority" label as we wouldn't hold the release up for it. The indirect tests are in now and that was a big chunk and Matt is dealing with the ALC ones. Let's put this as a maintenance task that might be best addressed as the Qt4 targets are removed where we can spot what files are just deleted.

martyngigg, 2021-01-27:
> From [scripts/test/CMakeLists.txt](https://github.com/mantidproject/mantid/blob/master/scripts/test/CMakeLists.txt#L30-L37):
>
>     * SANSBatchModeTest.py
>
>     * SANSCentreFinderTest.py
>
>     * [...]

I don't know much about these. @DavidFair Would you be able to comment at all?

DavidFair, 2021-01-28:
AFAIK these should be running in Qt4 and Qt5 as most of the other SANS GUI tests are and there shouldn't be anything preventing them running, however they predate me. Does this mean they aren't running on Qt 5 then?

Really they should be with the other SANS GUI tests rather than separate but that's a different problem of organisation.

peterfpeterson, 2021-01-28:
Look at the `CMakeLists.txt` that Martyn pointed out. Those tests are currently only run in qt4. I don't know if it is because they don't work under qt5 currently, but that file suggests they didn't always work with qt5. This may be a simple job of removing the `if (ENABLE_WORKBENCH)` and fixing what is broken.

martyngigg, 2021-01-29:
Thanks for checking @DavidFair. I'll have a quick look today at just putting them in the Qt5 bunch and hopefully things will run 🤞🏻

martyngigg, 2021-01-29:
It looks good. The pythonTSV test is the only Qt4-only one and that's a MantidPlot thing anyway.

martyngigg, 2021-03-02:
For me to keep track.

Items to look at:
  - qt/scientific_interfaces/CMakeLists.txt: Missing tests in the Qt5 list
  - qt/widgets/plotting: Missing Qt5 test target

Old sliceviewer has been completed removed so should no longer exist anyway.

PR #30824 (2021-04-26): Ensure all tests run under Qt5
**Description of work.**

During the removal of Qt4 targets some tests were accidentally removed from running. This reinstates the ones that are not for targets that will be removed. It also removes the general test target for ScientificInterfaces in favour of each library having its own test executable. To this end the reminaing Muon & all ISIS Reflectometry tests have moved to subfolders of their respective libraries.

I also fixed a warning in `qt/scientific_interfaces/ISISReflectometry/test/Options/OptionsDialogPresenterTest.h` that clang gave me about a defa [... cut]
closed 2021-04-26 by DavidFair