mantidproject/mantid #30451: Record individual function usage statistics

StephenSmith25, 2021-01-22:
It would be good to record the individual function usage statistics - At the moment we have circa. 130 functions, so it would be good if we could reduce the bloat by removing unused functions.

SpencerHowells, 2021-01-24:
@StephenSmith25 This needs to be done with care. Usage of functions depends on the science that has been done by the users which is unpredictable. Instrument scientist will need to be consulted.

StephenSmith25, 2021-01-24:
Not to worry, any actions taken will be carefully considered and I'll make sure the relevant parties are consulted. This was just motivated by having to write documentation this week for several undocumented functions, which nobody knew anything about - not even the scientific group in that area. I believe taking stock of what we have and what's being used will be a helpful statistic to have.

PR #30605 (2021-03-05): Track function usage statistics

**Description of work.**
This PR adds function usage tracking. This tracking is only performed when the function is about to be used in a fit. The alternative was to track the usage every time the function was created, however this resulted in a lot of usages which weren’t "true" usages.

**To test:**
To test this you'll probably need to build the usage reporter server and re-route mantid to your docker instance https://github.com/mantidproject/reports. This may be a bit of a pain, but if you need a hand setting it up please slack me.

- Code review
- With the  [... cut]
closed 2021-03-05 by gvardany