mantidproject/mantid #30464: Spectra Index error in indirect FqFit with EISF and Width

Harrietbrown, 2021-01-25:
Switching between Width and EISF in the add multiple dialogue in FqFit can produce a warning.

1. Open indirect, 
2. go to fqfit,
3. multiple input,
4. add workspace,
5. load file wioth both eisf and width, switch to eisf and add spectra. 
5. a warning pops up about spectra index

AnthonyLim23, 2021-04-28:
@Harrietbrown does this need to be fixed for this release?

martyngigg, 2021-04-29:
The assumption is this is not a priority for 6.1 but we can reassess later.

Harrietbrown, 2021-05-05:
This issue seems to have been fixed by another PR. I am going to close it.

Harrietbrown, 2021-05-07:
This is still an issue that should be fixed for release.

1.	open f(q) fit
2.	go to multiple input
3.	load data with both Width and EISF but only add the width
4.	change the function to something
5.	remove the spectra
6.	try to add the EISF data and you are unable as the list of entries is blank
7.	reloading the workspace will show the list of available eisf but adding gives an error related to the spectra index
8.	if you try and change the function after adding an EISF mantid crashes

martyngigg, 2021-05-10:
@Harrietbrown Is this for 6.1? Can we set the milestone if so?

Harrietbrown, 2021-05-10:
Yes thanks for reminding me

gemmaguest, 2021-05-14:
Fixed by #31423

PR #31423 (2021-05-14): FIx bugs with IndirectDataAnalysis f(q) fit add workspace Dialog
**Description of work.**

This PR fixes bugs in the indirect data analysis F(Q) fit that caused errors when:
1. Removing all spectra and then trying to add a new one.
2. Closing and reopening the add workspace dialogue leaving drop down menus empty.

**Report to:** @SpencerHowell

**To test:**

1. open f(q) fit
2. go to multiple input
3. load data with both Width and EISF but only add the width
4. close the dialog
5. reopen the dialoge it should be reset
6. close the dialog
7. remove the spectra you had add
8. Add the EISF data,  [... cut]
closed 2021-05-14 by gemmaguest