mantidproject/mantid #30474: Missing fields in SaveCanSAS1D and SaveNXcanSAS

gemmaguest, 2021-01-25:
### Expected behavior
SaveCanSAS1D and SaveNXcanSAS should include `sample_scatter_run` in the `SASprocess` block

### Actual behavior
`sample_scatter_run` is missing

### Steps to reproduce the behavior
You'll need the training data.
- Open the ISIS SANS interface
- Click `Load User File` and select `MaskFile.toml` from the `loqdemo` subdirectory in the training data.
- Click `Load Batch File` and select `batch_mode_reduction.csv`
- Ensure the default save directory is set to something sensible
- Ensure CanSAS and NXCanSAS are ticked under Save Options
- Select the first row in the table and click `Process Selected`
- Open the saved `first_time_HAB_1D_2.2_10.0.xml` file in a text editor and note that in the SASprocess bock the required field is missing
- Open `first_time_HAB_1D_2.2_10.0.h5` in HDFView and note that in SASprocess->sasnote the required field is missing

### Platforms affected
All

gemmaguest, 2021-01-25:
@smk78 It looks like `can_trans_run` is also missing from SaveNXcanSAS?

smk78, 2021-01-25:
Actually, it looks like can_trans_run is there, but under <sasprocess> top level and not <sasprocess\sasnote>.
[image]
[issue30474.zip](https://github.com/mantidproject/mantid/files/5869008/issue30474.zip)
I can't immediately think why they should be in different locations...

closed 2021-08-01 by stale[bot]