mantidproject/mantid #30492: Error when renaming workspace that has spectrum plotted

RichardWaiteSTFC, 2021-01-27:
### Expected behavior
The spectrum plotted would continue to be plotted and the legend would be updated (if possible).

### Actual behavior

```
Error occurred in handler:
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\workbench\plotting\figuremanager.py", line 51, in wrapper
    func(*args, **kwargs)
  File "C:\MantidNightlyInstall\bin\workbench\plotting\figuremanager.py", line 143, in renameHandle
    for ws_name, artists in ax.tracked_workspaces.items():
RuntimeError: dictionary keys changed during iteration
```

### Steps to reproduce the behavior
(1) Plot a spectrum
(2) Rename workspace

### Platforms affected

StephenSmith25, 2021-01-27:
I think this is the same bug as here: #29947

RichardWaiteSTFC, 2021-01-27:
Yep you're right, sorry! I'll close this issue.

closed 2021-01-27 by RichardWaiteSTFC