mantidproject/mantid #30502: CalculateMonteCarloAbsorption crash

DanielMurphy22, 2021-01-27:
**Original reporter:** Oliver Dicks

### Actual behavior
Crash when running CalculateMonteCarloAbsorption
### Steps to reproduce the behavior
Load LET72350.nxspe or LET72352.nxspe from the ISIS data archive into workbench
Search for the algorithm [CalculateMonteCarloAbsorption](https://docs.mantidproject.org/nightly/algorithms/CalculateMonteCarloAbsorption-v1.html), set the shape type to **Annulus** and run --> hard crash.
### Platforms affected
Seen on IDAaaS, 5.1.20210126.813

SpencerHowells, 2021-01-27:
If this is old data then it probably does not have the parameter efixed set.

SpencerHowells, 2021-01-27:
The algorithm requires that input workspace to be in wavelength whereas nxspe is in energy. So goodness knows what happens when it’s run! User should use the interface Indirect>Corrections after setting efixed.

DannyHindson, 2021-01-28:
I had a look at this issue. There is a bug in the validation code for the algorithm that causes the crash. The validation finds a genuine problem (which is that there isn't an efixed value in the workspace) but then the exiting of the validation routine isn't done in a v nice way and the code falls over in the handling of this error.

So I can fix it. The algorithm still won't work though because the efixed value is missing. There isn't a totally solid convention for where to store the efixed value for direct instruments but it's normally in the sample logs with a name "Ei" I think

I just wanted to check one detail - on the description for the issue it suggests loading one of two .nxspe files. What does the .nxspe extension mean? I could only find two .nxs files on the archive. I was wondering if the .nxspe files might have the efixed value in them if they've been processed a bit more

Finally, the algorithm is now deprecated and has been replaced by PaalmanPingsMonteCarloAbsorption so even when if\when we do get this to work we may want to switch over to that (which doesn't suffer from the same bug - I've just checked)

DannyHindson, 2021-01-28:
Just noted your comments Spencer - hadn't refreshed the page when I applied mine

SpencerHowells, 2021-01-30:
I’ve tested on several Direct instruments which have efixed defined & it seems to work well. I used the Indirect>Corrections interface for both Calculate & Apply.

martyngigg, 2021-05-13:
I've been slow at cleaning up 6.1S3 and this is the only thing left. Just checking if it is still a valid issue?

DanielMurphy22, 2021-05-13:
I think this should be closed, the issue is resolved for the user.

closed 2021-05-13 by DanielMurphy22