mantidproject/mantid #30503: Add string only keys to more Singletons

peterfpeterson, 2021-01-27:
As a follow on to #30487, keys to strings in more singletons. This is for things in [kernel](https://github.com/mantidproject/mantid/pull/30487/files#diff-1557027ce2591fd3c258d8a6b36ec251c42df9c26e55a22e876f2c87ec44cc48L61-L64) and [api](https://github.com/mantidproject/mantid/pull/30487/files#diff-51cbb5f08b393d04bd9b83cef2209372e5061e588398a687f181c2fe3fce82eeR26-R34).

`api/_aliases.py`:
* `AnalysisDataService` - already done in #30487
* `AlgorithmFactory` - yes
* `AlgorithmManager` - maybe
* `FileFinder` - maybe
* `FileLoaderRegistry` - maybe
* `FrameworkManager` - maybe
* `FunctionFactory` - maybe
* `WorkspaceFactory` - maybe
* `CatalogManager` - maybe

`kernel/_aliases.py`:
* `UsageService` - maybe
* `ConfigService` - yes
* `PropertyManagerDataService` - maybe
* `UnitFactory` - maybe

martyngigg, 2021-04-28:
I guess we're probably not going to do do this for 6.1?

martyngigg, 2022-01-05:
@peterfpeterson I'm going to push this back again. Let me know if you'd like to work on it.

PR #33379 (2022-02-01): Force keys to some singletons to be strings
This is leveraging work done in #30487 to simplify calling patterns.
See that chunk of work for better context as to what this is doing.

**To test:**

A code review is sufficient.

Fixes #30503

*This does not require release notes* because it is a minor usability improvement

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.ma [... cut]
commit d663d731 (2022-02-01): Force keys to some singletons to be strings

This is leveraging work done in #30487 to simplify calling patterns.
See that chunk of work for better context as to what this is doing.

Fixes #30503
commit 719fecae (2022-02-01): Force keys to some singletons to be strings

This is leveraging work done in #30487 to simplify calling patterns.
See that chunk of work for better context as to what this is doing.

Fixes #30503
closed 2022-02-01 by jmborr