mantidproject/mantid #30504: MacOS version crashes

AndreiSavici, 2021-01-27:
**Original reporter:** SNS/granroth

### Expected behavior
Be able to start MantidWorkbench

### Actual behavior
Crash

### Steps to reproduce the behavior
`The release crashes when I try to run it although if I run mantid python it seems to be in Python 3.8.  Does Mantid need 3.6?   If I run the nightly it complains about not being able to find a library in /usr/local/boost.  There is no /usr/local/boost directory.  I am  on mojave.`

Comment from @quantumsteve
` I wonder if the name change between MantidWorkbench and MantidWorkbenchNightly is causing MantidWorkbenchNightly to not be fixed up in packaging `
https://github.com/mantidproject/mantid/blob/c7dc6157d39bbc8fe94fa60828333707cc9c2399/installers/MacInstaller/make_package.rb#L593
### Platforms affected
MacOS

martyngigg, 2021-02-04:
We're getting pretty close to the release date next week. Do we think we can get this in?

quantumsteve, 2021-02-04:
@martyngigg I created a PR using Filepath.basename instead of `MantidWorkbench` or `MantidPlot`. Hopefully that's sufficient to address the issue.

granrothge, 2021-02-05:
@quantumsteve @AndreiSavici The behavior is now consistent with the current distribution. When I run this, the app icon jumps for awhile and then disappears.  If I go into the application folder and try to run it I get. 

`/Applications/MantidWorkbenchNightly.app/Contents/MacOS/MantidWorkbenchNightly ; exit;
Traceback (most recent call last):
  File "/Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/workbench/app/main.py", line 55, in <module>
    main()
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/workbench/app/main.py", line 46, in main
    start(options)
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/workbench/app/main.py", line 50, in start
    from workbench.app.start import start
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/workbench/app/start.py", line 29, in <module>
    from workbench.plugins.exception_handler import exception_logger  # noqa
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/workbench/plugins/exception_handler/__init__.py", line 12, in <module>
    from mantidqt.dialogs.errorreports.report import CrashReportPage
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantidqt/dialogs/errorreports/report.py", line 30, in <module>
    class CrashReportPage(ErrorReportUIBase, ErrorReportUI):
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantidqt/dialogs/errorreports/report.py", line 34, in CrashReportPage
    interface_manager = InterfaceManager()
Exception: unknown
Traceback (most recent call last):
  File "/Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantidqt/dialogs/errorreports/main.py", line 16, in <module>
    from mantidqt.dialogs.errorreports.report import CrashReportPage
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantidqt/dialogs/errorreports/report.py", line 30, in <module>
    class CrashReportPage(ErrorReportUIBase, ErrorReportUI):
  File "/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantidqt/dialogs/errorreports/report.py", line 34, in CrashReportPage
    interface_manager = InterfaceManager()
Exception: unknown
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
`

quantumsteve, 2021-02-05:
@granrothge Can you start `mantidpython` from the command line and import mantid?

```
cd /Applications/MantidWorkbenchNightly.app/Contents/MacOS
./mantidpython
from mantid.simpleapi import *
```

martyngigg, 2021-02-08:
Fixed by #30585

martyngigg, 2021-02-08:
My script for autoclosing issues caught this. Reopening until we understand more.

DanielMurphy22, 2021-02-08:
The current nightly `mantidnightly-5.1.20210205.1617` opens fine for me on Mac Catalina.

martyngigg, 2021-02-08:
Fixed by #30585

closed 2021-02-08 by martyngigg

martyngigg, 2021-02-08:
I can confirm this works for me to. I also checked by running `mv /usr/local/opt /usr/local/optx` to verify there were no Homebrew links left.

quantumsteve, 2021-02-08:
We narrowed it down to settings in @granrothge 's Mantid.user.properties file. After renaming that file, MantidWorkbench started correctly.

DanielMurphy22, 2021-02-08:
@quantumsteve If you can find the exact entry that caused the problem, that'd be great! If you don't have the time I'd be happy to have a quick look at the user.properties file 🙂

granrothge, 2021-02-08:
the offending line was `instrumentDefinition.directory=./instrument`. It is now `instrumentDefinition.directory=../instrument`. Maybe it should fail a little more gracefully?

I do have `UpdateInstrumentDefinitions.OnStartup=0`

quantumsteve, 2021-02-08:
@DanielMurphy22 Here's the error in MantidPython after adding `instrumentDefinition.directory=./instrument` to my Mantid.user.properties file.
```
svh@mac102014 MacOS % ./mantidpython 
Python 3.8.3 (default, May 27 2020, 20:53:50) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.14.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import mantid.simpleapi
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-1-ecb2fe53065b> in <module>
----> 1 import mantid.simpleapi

/Applications/MantidWorkbenchNightly.app/Contents/MacOS/mantid/simpleapi.py in <module>
   1267 #   - loads the python plugins and create new algorithm functions
   1268 if not _api.FrameworkManagerImpl.hasInstance():
-> 1269     _api.FrameworkManagerImpl.Instance()
   1270 _translate()
   1271 

RuntimeError: Failed to find "Facilities.xml". Searched in /Users/svh/.mantid/instrument/

In [2]:
```

quantumsteve, 2021-02-08:
Here's the line where the exception is thrown.

https://github.com/mantidproject/mantid/blob/6e9bd1cccdafcc2c65f4ccec7ec82a152b2fbb6b/Framework/Kernel/src/ConfigService.cpp#L1678

martyngigg, 2021-02-08:
@DanielMurphy22 Didn't we create an issue about improving feedback to users in scenarios like this where users essentially see nothing happen.

DanielMurphy22, 2021-02-08:
> @DanielMurphy22 Didn't we create an issue about improving feedback to users in scenarios like this where users essentially see nothing happen.

I thought we had. Struggling to find anything. This is the closest I've found https://github.com/mantidproject/mantid/issues/27462

AndreiSavici, 2021-02-08:
I will open a new issue

PR #30585 (2021-02-05): Fix nightly MacOS app bundle

**Description of work.**

Use Pathname.basename to get executable name

**Report to:** @AndreiSavici @granrothge 

**To test:**

Download MacOS .app bundle and try to start MantidWorkbench.

Fixes #30504. 

*This does not require release notes* because only affects nightly builds

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject [... cut]