mantidproject/mantid #30514: Feedback on the new ISIS SANS TOML User Files

smk78, 2021-01-28:
### Background
Mantid v5.1.0 introduced a beta of a new TOML-based User File for the ISIS SANS Interface. At the present time this can be used in parallel with the existing, historic, plain text-based, '_COLETTE_'-format User File.

This ticket is intended to capture feedback and issues arising during User acceptance testing of the TOML-format.

### Examples
This is an example of a User File in the old format: 
[USER_LOQ_202B_M3_Xpress_Campana_12mm_Changer_MAIN_LIN.txt](https://github.com/mantidproject/mantid/files/5888890/USER_LOQ_202B_M3_Xpress_Campana_12mm_Changer_MAIN_LIN.txt)

This is an example of the same User File in the TOML format (v0):
[USER_LOQ_202B_M3_Xpress_Campana_12mm_Changer_MAIN_LIN_toml.txt](https://github.com/mantidproject/mantid/files/5888904/USER_LOQ_202B_M3_Xpress_Campana_12mm_Changer_MAIN_LIN_toml.txt)

smk78, 2021-01-28:
Q1. Is indenting required, or just a nicety? I believe it is not a requirement of the TOML standard, but what about the parser in Mantid?

Q2. Similarly, do strings have to be enclosed by speech quotes (") or is a single quote (') also acceptable (cf, Python)?

Q3. Why is _collimation_length_ being specified as a string?

Q4. A semantic maybe, but should wavelength binning be under _[binning.wavelength]_?

**COMMENT: The new format allows "RangeLin" and "RangeLog" for wavelength binning to be specified in a User File!**

**COMMENT: The new format does not support compatibility mode! (Though this is being withdrawn as a reduction option in a future Mantid release anyway)**

Q5. _[detector.configuration]_ | _selected_detector_ allows "LAB" (low angle bank) and "HAB" (high angle bank). But elsewhere the new format uses "rear" and "front" detector. Would it be cleaner to just standardise on one form?

Q6. _[detector]_ contains elements which are _[detector.calibration.?]_. Would these be better called _[detector.correction.?]_?

Q7. Some of the elements under _[mask]_ seem a little confusing: _beamstop_shadow_ and _mask_pixels_ are spatial masks, so why not put them under _[mask.spatial.rear]_? (NB: some pixels might, of course, be on a front detector)

Q8. Similarly, as there is _[mask.time]_ | _tof_ = [{},{}} do we really need the separate time mask _[mask]_ | _prompt_peak_ too?

**COMMENT: There are no entries for _[normalisation]_ or _[transmission]_ in https://docs.mantidproject.org/nightly/techniques/SANS-Toml-v1.html**

**COMMENT: I am confused about what the difference is between _[instrument.configuration]_ | _norm_monitor_ and _[normalisation]_ | _selected_monitor_ and, similarly, between _[instrument.configuration]_ | _trans_monitor_ and _[transmission]_ | _selected_monitor_.**

DavidFair, 2021-01-29:
> Q1. Is indenting required, or just a nicety? I believe it is not a requirement of the TOML standard, but what about the parser in Mantid?

A nicety - we use a external TOML parser which internally turns stuff into a Python dictionary, so were standard complient

> Q2. Similarly, do strings have to be enclosed by speech quotes (") or is a single quote (') also acceptable (cf, Python)?

If I recall correctly, both are accepted. It's just conventional in the programming world that single quotes are used for single characters (e.g. '0') and double for >2 chars (e.g. "foo")

> Q3. Why is collimation_length being specified as a string?

This is likely a typo on my part, both "0.1" and 0.1 would be accepted (although the string is more likely to break things). It depends if we do a `float(collimation_length)` later on if this works (hence the mistake was accepted)

> Q4. A semantic maybe, but should wavelength binning be under [binning.wavelength]?

Can you compile semantic changes like this from the group, this is the sort of stuff I want to nail down in v1 :+1: 

> COMMENT: The new format allows "RangeLin" and "RangeLog" for wavelength binning to be specified in a User File!

Yeah not sure why the old parser didn't support it, is it working as you expect?

> COMMENT: The new format does not support compatibility mode! (Though this is being withdrawn as a reduction option in a future Mantid release anyway)

We run with [compatibility_mode = True by default](https://github.com/mantidproject/mantid/blob/97905da9f4d2519b916f17857d63ceb32292b43e/scripts/SANS/sans/state/StateObjects/StateCompatibility.py#L31) hence I decided to drop the option as compatibility mode off is broken AFAIK and nobody has noticed/been bothered by it....

> Q5. [detector.configuration] | selected_detector allows "LAB" (low angle bank) and "HAB" (high angle bank). But elsewhere the new format uses "rear" and "front" detector. Would it be cleaner to just standardise on one form?

Completely agree - I'll leave it up to the group to decide which "one true syntax" we will adopt :wink: 

> Q6. [detector] contains elements which are [detector.calibration.?]. Would these be better called [detector.correction.?]?

See above about compiling semantic changes, I'll add it to the list

> Q7. Some of the elements under [mask] seem a little confusing: beamstop_shadow and mask_pixels are spatial masks, so why not put them under [mask.spatial.rear]? (NB: some pixels might, of course, be on a front detector)

Ditto

> Q8. Similarly, as there is [mask.time] | tof = [{},{}} do we really need the separate time mask [mask] | prompt_peak too?

Internally we have a separate set of fields to track prompt peak, although I assume (without checking) they do the same. I imagine it was originally put in to make it clear to users why we were masking that TOF range

> COMMENT: There are no entries for [normalisation] or [transmission] in https://docs.mantidproject.org/nightly/techniques/SANS-Toml-v1.html

Huh that is a massive oversight from me copying the template, I'll fix that when I get back from rotation

> COMMENT: I am confused about what the difference is between [instrument.configuration] | norm_monitor and [normalisation] | selected_monitor and, similarly, between [instrument.configuration] | trans_monitor and [transmission] | selected_monitor.

So this is weirdness in the state objects I didn't have time to dig

`inst.config.norm_monitor` and `inst.config.transmission` gets assigned to a `StateCalculateTransmission` objects incident and transmission monitor fields respectively

`norm.selected_monitor` uses the name provided to go to `norm.monitor_name` to find the spectrum number in that section to then assign to the same `StateCalculateTransmission` objects transmission monitor field. So this is likely a bug where we have the same attribute represented by two fields. I need to dig a little more to be 100% sure though.

DavidFair, 2021-01-29:
Semantic Feedback so far:

- [ ] should wavelength binning be under [binning.wavelength]?
- [ ] standardise on one form on front/rear or lab/hab
- [ ]  [detector] contains elements which are [detector.calibration.?]. Would these be better called [detector.correction.?]
- [ ] Some of the elements under [mask] seem a little confusing: beamstop_shadow and mask_pixels are spatial masks, so why not put them under [mask.spatial.rear]? (NB: some pixels might, of course, be on a front detector)
- [ ] (Possibly needs discussion) There is [mask.time] | tof = [{},{}} do we really need the separate time mask [mask] | prompt_peak too?

DavidFair, 2021-01-29:
Bugs / Fixes:
- [ ] There are no entries for [normalisation] or [transmission] in https://docs.mantidproject.org/nightly/techniques/SANS-Toml-v1.html
- [ ] (see above comment) I am confused about what the difference is between [instrument.configuration] | norm_monitor and [normalisation] | selected_monitor and, similarly, between [instrument.configuration] | trans_monitor and [transmission] | selected_monitor.

smk78, 2021-02-23:
Q9. The new format allows the radius limits to be specified, eg:
```
[detector]
  radius_limit = {min = 0.035, max = 0.419}
```
but where is the integration ring width specified?

smk78, 2021-04-26:
Q10. An observation, not a question. But in testing Nighty 6.0.20210422.2105 on LOQ data it was observed that the established .TXT based User files and the .TOML based User files were cross-setting the LAB & HAB beam centre coordinates:

TXT (the correct settings)
[image]

TOML
[image]

Unfortunately this renders output from TOML-based User files totally incorrect.

smk78, 2021-04-27:
Q10. What is the required spelling of normalisation/normalization? Even the new docs are confused!
[image]
Personally, I think the parser should embrace both spellings.

smk78, 2021-04-27:
11. The following User Files are intended to be equivalent, but actually generate data which is different. It looks like it is something to do with the transmissions.
[user_files.zip](https://github.com/mantidproject/mantid/files/6383445/user_files.zip)

This needs looking in to.
So the as calculated transmissions look like this:
[image]
The Polynomial3 fits look like this:
[image]
And the reduced data look like this:
[image]

DavidFair, 2021-07-26:
> 1. The following User Files are intended to be equivalent, but actually generate data which is different. It looks like it is something to do with the transmissions.
> [snip]
> However, this would seem to imply that when using M4 no background is currently being subtracted from the M4 spectrum?

- [x] This turned out to be a missing moderator command, which probably because I forgot to add said command to the TOML documents

dehoni, 2021-07-27:
For LARMOR, here is an example user file for a standard sample changer experiment:
[USER_Larmor_211B_Changer_r57814.txt](https://github.com/mantidproject/mantid/files/6884814/USER_Larmor_211B_Changer_r57814.txt)
An attempt to TOMLize this file:
[USER_Larmor_211B_Changer_TOML.toml.txt](https://github.com/mantidproject/mantid/files/6884824/USER_Larmor_211B_Changer_TOML.toml.txt)

Playing with this file some questions and remarks arised:
Q1: The [q_resolution] parameters should be in m and not in mm like all other distances.
Q2: Loading the TOML files via “Load user file” of the SANS reduction GUI works if no file format is selected in the lower right box. If selecting .toml as file format, no files are visible.
Q3: [instrument.configuration]> sample_aperture_diameter is not listed in the SANS TOML file description.
Q4: For consistency, “sample_aperture_diameter” (under [instrument.configuration]) and the aperture sizes H and W (appearing under [q_resolution] ) should be collected in the same structure either [instrument.configuration] or [q_resolution].
Q5: The notion rear and front detector might be confusing. I understand that it is inspired from LOQ and SANS2D with two physically separated detector banks. However, for Larmor and Zoom having only one main detector, it would be meaningless. 
Q6: For [binning.1d_reduction] no type “Log/Linear” (like for 2D) is defined. A negative step indicating Log is very non-intuitive for a user. Could [binning.1d_reduction] also have explicit “start = xx” and “stop=xx” like for other parameters to enhance the readability for users.
Q7: Could “[gravity]  enabled = true” put into [instrument], i.e. close to “gravity extra length”. Otherwise it tends to float around in the file.
Q8: The detector radius L/R needs some more explanation. In the user files it has as third parameterthe ring width values always, but “[detector]  radius_limit” only considers 2.

pzt29813, 2021-08-06:
I made a mistake in the TOML file for [detector.calibration.direct] where I left the units in mm rather than converting to m. I did not notice the issue on the rear but only on the front as altering rear_x from 2.5 to 0.0025 makes no difference to the data reduction...I assume this is due to the reduction being carried out from the beam centre calculated from the rear detector? I guess we should check.

pzt29813, 2021-08-06:
I also observed the following when reducing 'All' on Sans2d.  The following needs to be used when carrying out a reduction on both detectors (but not merging them):

[detector.configuration]
  selected_detector = "All"
  front_centre = {x=0.10731, y=-0.08203}
  rear_centre = {x=0.10731, y=-0.08203}
  #front_scale = 1.0
  rear_scale = 0.02415

This is annoying as in the old user files we only needed to set one centre for Sans2d (I do not need to use a front_scale though so will remove this from the file).

It is also difficult to bug track down this as the gui does not change to reflect this difference i.e. the gui always looks like this:

[image]

In the end I did a show history on the nxs file and then looked through the SANSState for each reduction and found this for the HAB (front) when the front_centre is #d out:

"sample_centre_po1": 0.0, "sample centre_pos2" = 0.0

whereas it should have been (to be consistent with the txt and I therefore assume give the right answer!):

"sample_centre_po1": 0.10731, "sample centre_pos2" = -0.08203

i.e. the same as the LAB (rear)

PR #32081 (2021-08-20): Toml v1 feedback
**Description of work.**

*We're not at the point where the group is comfortable committing completely to V1 yet, so we will leave version as V0 (despite having breaking changes during this beta testing phase), since there is a limited number of files we've converted.*

List of changes:
- Fixed TomlValidationErrors causing a Sad Mantid
- List of changes to TOML schema as documented at the top of `Sans-Toml-v1.rst`

@smk78 

**To test:**
- Download the updated [training test data](https://sourceforge.net/projects/mantid/files/Sample%20Data/TrainingCourseData.zip/download)
- Fin [... cut]
closed 2021-08-20 by martyngigg