mantidproject/mantid #30525: Muon Analysis crash after switching around start and end X

robertapplin, 2021-01-29:
**Original reporter:** James Lorde

### Expected behavior
A fit should be performed ok.

### Actual behavior
There is a crash in James' demonstration. For me it uses the old start and end X so it says the x-range is invalid.

### Steps to reproduce the behavior
1. Go to Muon Analysis
2. Select HIFI instrument and `Load Current Run`
3. Go to `Fitting Tab`
4. Add functions `Chebyshev` and `ExpDecayOsc`
5. Set `EndX` = 0.0, `StartX` = 15.0
6. Set the `n` parameter in Chebyshev to 2
7. Set the Frequency parameter in ExpDecayOsc to 0.6
8. Click `Fit`. An invalid x-range warning appears correctly.
9. Switch the `EndX` = 15.0 and `StartX` = 0.0
10. Click `Fit` again and crash, or same warning appears!

### Platforms affected
All I assume

AnthonyLim23, 2021-02-01:
@Matt-Cumber I thought you made it automatically switch the values if min was greater than max?

Matt-Cumber, 2021-02-01:
> @Matt-Cumber I thought you made it automatically switch the values if min was greater than max?

I have just been trying this on master, the startx and endx switch for me and I don't get any warnings

robertapplin, 2021-02-01:
Have you tried release-next? I get this warning 
[image]

robertapplin, 2021-02-01:
The second time the warning shouldn't appear because I have switched them around to be correct

robertapplin, 2021-02-01:
@Matt-Cumber This is referring to the `StartX` and `EndX` within the Chebyshev function itself, not the Time Start or Time End

Matt-Cumber, 2021-02-01:
> @Matt-Cumber This is referring to the `StartX` and `EndX` within the Chebyshev function itself, not the Time Start or Time End

Ah right that will be why then @AnthonyLim23

AnthonyLim23, 2021-02-01:
ah ok, so that is an issue with the fitting function then?

robertapplin, 2021-02-01:
I think the issue is the Chebychev function is stored in the fitting tab code and is not being updated when start/end x is changed. So it is always using the original start/end x

Matt-Cumber, 2021-02-08:
Fixed by #30555

PR #30555 (2021-02-04): Fix StartX EndX Function Parameters Not Updating In MA
**Description of work.**
I've added a final get function before the fit is performed to make sure the most recent data is being used

**To test:**

1. Go to Muon Analysis
2. Select HIFI instrument and run 83934
3. Go to Fitting Tab
4. Add functions Chebyshev and ExpDecayOsc
5. Set EndX = 0.0, StartX = 15.0
6. Set the n parameter in Chebyshev to 2
7. Set the Frequency parameter in ExpDecayOsc to 0.6
8. Click Fit. An invalid x-range warning appears correctly.
9. Switch the EndX = 15.0 and StartX = 0.0
10. Click Fit again, should work this ti [... cut]
closed 2021-02-08 by Matt-Cumber