mantidproject/mantid #30526: Muon Analysis fitted curve has same colour as raw data

robertapplin, 2021-01-29:
**Original reporter:** James Lorde

### Expected behavior
The fitted line should be plotted in a different colour to the raw data.

### Actual behavior
The fitting line is plotted with the same colour as the raw data, making it difficult to see.

### Steps to reproduce the behavior
1. Muon Analysis interface
2. HIFI instrument. Load runs 175833-175843
3. Go to Fitting tab
4. Add functions `Linear Background` and `ExpDecayOsc`
5. Change parameter frequency to 0.6
6. Click `Fit`. It does the fit but the fitted data is plotted in the same colour.

### Platforms affected
Found on windows

AnthonyLim23, 2021-04-28:
I think this is fixed

Matt-Cumber, 2021-05-07:
Re-opening this issue as it still exists and can be reliably reproduce. Loading EMU20883-93 and doing a fit causes the fit line to be the same colour as well as MUSR62260-62. Looks to be related to there being more than 10 plotted lines at once resulting in the colour queue to be reset

martyngigg, 2021-05-12:
Fixed by #31389

PR #31389 (2021-05-12): Muon Analysis Fix Fit Line Being Same Color As Data Line
**Description of work.**
Added a check before adding colors back to the color queue when lines are removed

**To test:**
Open Muon Analysis
You need to load data so that there are more than 10 plots (10 is the current number of colors)
An example is EMU20883-20893 or MUSR62260-62262
You should see that the colors are repeated for some lines 
Go to the fitting tab
Do a fit
The fit line should not be the same color as the raw data line
Check stepping through workspaces on the fitting tab keeps the raw data line as blue and the fit line ora [... cut]
closed 2021-05-12 by martyngigg