mantidproject/mantid #30535: Need to add `refl1d` and `bumps` as Mantid dependencies

marshallmcdonnell, 2021-01-29:
Carry over from previous issue:
https://github.com/mantidproject/mantid/issues/28626

Specifically, the [following comment](https://github.com/mantidproject/mantid/issues/28626#issuecomment-637104901) was made at the end of closing the issue:
> Still waiting for a refl1d package release that will not included wxpython.
> In the mean time, while we wait:
>     Submit the current work as a PR with check that refl1d is installed. Error if not installed and disabled tests if not installed
>     Once refl1d has the new release, will try to get it added as a dependency of Mantid.
> If we can get refl1d as a proper dependency of Mantid, will then refactor to remove the need to check that it is installed in the algorithm, in the test, and change the release note (if same release). Can open a separate issue, will post here for cross-reference.

The [v0.8.13 release](https://github.com/reflectometry/refl1d/releases/tag/v0.8.13) of [refl1d](https://github.com/reflectometry/refl1d) has removed `wxpython` dependency

Notify: @mdoucet @peterfpeterson @mabrahamdevops

martyngigg, 2021-04-28:
I'm just cleaning up issues for 6.1 now we're in feature freeze - should this be bumped to 6.2?

martyngigg, 2022-05-05:
We keep rolling this over. Do we actually plan on doing anything with it?

closed 2023-07-30 by stale[bot]