mantidproject/mantid #30558: Muon differences clean up and tests

AnthonyLim23, 2021-02-02:
This is to clean up the code for https://github.com/mantidproject/mantid/pull/30465 and adding tests. 

The code should have the MuonDiff moved into its own file
In MuonDiff we should rename the properties as positive and negative
Add tests

AnthonyLim23, 2021-02-03:
in workspace naming change the globals to be all caps

AnthonyLim23, 2021-02-05:
we should add tests like  this https://github.com/mantidproject/mantid/pull/30465/files#diff-f9feb221c2393ddbc00bc7c1ce81d86ba81c42c4e206faf50aed97805d1738f0L219 but instead of checking that the pair has been deleted. We should check that a warning has been produced.

AnthonyLim23, 2021-02-05:
we should remove the code needed for deleting pairs that depend on groups that have been deleted.

AnthonyLim23, 2021-02-05:
If you load a multi period XML into single period that the pair difference table does not colour highlight.

I have had a quick go and its related to the pair's not recording their periods. However, adding it to the xml reader causes it to crash.

AnthonyLim23, 2021-02-08:
use this https://github.com/mantidproject/mantid/pull/30465/files#diff-4199fe7e184523c3d1bb81b1887938ce24c974d708f5e870cf2fb89a07773816R169 as the return call on line https://github.com/mantidproject/mantid/pull/30465/files#diff-a511780dfd8943cfb54bfa7587743c76a6e37eb627cd420575a9ee12139121e6R99

AnthonyLim23, 2021-02-08:
we should talk to the muon group about how to handle changing between multi and single period runs. I like a warning.

AnthonyLim23, 2021-02-09:
feel free to split this up into smaller issues and PR's

Matt-Cumber, 2021-04-01:
Fixed by #30687 and #30732

closed 2021-04-01 by Matt-Cumber