mantidproject/mantid #30575: Add translation to change the integration centre of ellipsoids in IntegratePeaksMD

RichardWaiteSTFC, 2021-02-03:
The one-pass algorithm that calculates the covariance of the peaks also calculates the centroid - if the ellipsoidal integration was centred on the centroid it would better capture the peak tail. This option would also require changes to ellipsoid plotting in sliceviewer.

There do exist algorithms that calculate the centroid (e.g. `CentroidPeaksMD`) but there are two drawbacks to doing this
1. It changes the index of the peak - the peak centre will not coincide with the centroid as the peak is asymmetric - this way we preserve the index and get the intensity on the same peaks workspace with only one algorithm.
2. Such algorithms don't currently support background subtraction (though this could be added)

DannyHindson, 2021-03-15:
@RichardWaiteSTFC - just clearing out the old 6.1 S1 milestone. Can this issue be closed now? Looks like the PR got closed but the issue didn't also autoclose

RichardWaiteSTFC, 2021-03-15:
Yes good catch thanks - I will close it!

closed 2021-03-15 by RichardWaiteSTFC