mantidproject/mantid #30592: Drill hot commissioning

gvardany, 2021-02-05:
- [ ] Put forward a way to have initial state for the columns: present, folded, absent. 
- [ ] Make FluxRuns and AbsorberTransmissionRuns absent by default. It should be possible to bring them back from the context menu though.
- [ ] Do not pop-up the dialog asking `Do you want to save` if the table is empty.
- [ ] Check the serdon path on linux, should be /net4/ and not /net/

gui-co, 2021-02-24:
* [ ] allow loading of a rundex file that has no extension. For the moment, if the name of the file does not contain an extension, it is not displayed by the QFileDialog even with the "All files" option

gvardany, 2021-03-04:
- [ ] Implement possibility of typed columns. Particularly a checkbox for boolean and combo-box for choices will be very useful. For example, for the keep2D flag, it should be checkbox, rather than a lengthy string in custom options.
- [ ] When working on a table that is already saved to a rundex file, put the name of that file in the title of the main window of drill.

gvardany, 2021-03-22:
- [ ] Add another special column "Comment". This should not be used in any processing, but should be saved/loaded with the rundex. Users can freely type anything in it.