mantidproject/mantid #30606: Integration in PDFFourierTransform not treating start\end point correctly

DannyHindson, 2021-02-08:
It looks to me as though the q limits on the main integration loop in PDFFourierTransform2.cpp (and PDFFourierTransform.cpp) aren’t managed properly (this is the double loop in exec() at around line 518). Specifically:

-	Xmin_index is generally set to 1. This is fine for the calculations that depend only on the X values but it seems wrong for any calculations that depend on the Y values eg 0th value in inputY is skipped. Perhaps the code should be using inputY [inXindex-1] on the line where fs is incremented?
-	the value of Xmax_index appears to be one too small (this is calculated higher up in the unit)
-	the termination condition of the for loop looks to me as though it should be <= rather than <. This and the previous point mean the last two X bin edges are not processed

Then one final point that is lower priority. The calculation should possibly be using the x value of the bin centre during the calculation – at the moment it defines x as the right hand bin edge I think. But this is more of a design choice and not a bug unlike the points I've mentioned above.

```
for (size_t outXIndex = 0; outXIndex < sizer; outXIndex++) {
    const double outX = outputX[outXIndex];
    const double outXFac = corr / (outX * outX * outX);

    double fs = 0;
    double error = 0;
    for (size_t inXIndex = Xmin_index; inXIndex < Xmax_index; inXIndex++) {
      const double inX1 = inputX[inXIndex];
      const double inX2 = inputX[inXIndex + 1];
      const double sinx1 = sin(inX1 * outX) - inX1 * outX * cos(inX1 * outX);
      const double sinx2 = sin(inX2 * outX) - inX2 * outX * cos(inX2 * outX);
      double sinus = sinx2 - sinx1;

      // multiply by filter function sin(q*pi/qmax)/(q*pi/qmax)
      if (filter && inX1 != 0) {
        const double lorchKernel = inX1 * M_PI / inputX[Xmax_index];
        sinus *= sin(lorchKernel) / lorchKernel;
      }
      fs += sinus * inputY[inXIndex];

      error += (sinus * inputDY[inXIndex]) * (sinus * inputDY[inXIndex]);
    }

    // put the information into the output
    outputY[outXIndex] = fs * outXFac;
    outputE[outXIndex] = sqrt(error) * outXFac;
  }

DanielMurphy22, 2021-11-08:
Just my initial thoughts

> * Xmin_index is generally set to 1. This is fine for the calculations that depend only on the X values but it seems wrong for any calculations that depend on the Y values eg 0th value in inputY is skipped. Perhaps the code should be using inputY [inXindex-1] on the line where fs is incremented?

If Xminindex can be 0 then a little worried to use `inputY [inXindex-1]`. But this does work for the code snippet on the algorithm docs page.

> * the value of Xmax_index appears to be one too small (this is calculated higher up in the unit)

At a quick glance I wonder if this is due to this line (currently L206 on main) in determineMaxIndex()
`size_t first_normal_index = Y.size() - std::distance(Y.rbegin(), back_iter) - 1;`

> * the termination condition of the for loop looks to me as though it should be <= rather than <. This and the previous point mean the last two X bin edges are not processed

Yep I agree

> Then one final point that is lower priority. The calculation should possibly be using the x value of the bin centre during the calculation – at the moment it defines x as the right hand bin edge I think. But this is more of a design choice and not a bug unlike the points I've mentioned above.

Is the right bin edge used as a result using `inX2`? Not sure I totally follow this.

DannyHindson, 2021-11-09:
I think my comments on this issue were made at a point where Harriet was writing v2 of the algorithm so some of my points apply only to one of the versions. Here's a few answers to your questions:

1) Yes you're right. Would be safer to change determineMinIndex so that Xmin_index = 0 rather than use inXindex-1 as an index
I think the reason that Xmin_index is currently forced to be no less than 1 comes from the v1 algorithm (when the variable was called qmin_index). In the v1 algorithm the loop uses qmin_index and qmin_index -1 so I don't think there was a problem. In v2 however we use Xmin_index and Xmin_index + 1 so we're missing the first bin

2) Xmax_index is only too small if you use histogram data as input (and unfortunately the doc example uses point data - which isn't ideal because the algorithm outputs a warning saying it's not tested on point data). If you add a line `yy = np.delete(yy,-1)` to the doc example you will see that Xmax_index is set to 98. Your suggestion that it's the line where `first_normal_index` is set is the culprit. This line assumes the number of x values is the same as the number of y values.

3) I think my comment about the termination of the loop (the < vs <= issue) was only a problem in v1 of the algorithm where it looked backwards to find the previous bin edge ie used qmin_index and qmin_index -1

4) My final "low priority" comment was again targetted at the v1 algorithm where it read the y value from the right hand bin edge (from qmin_index). In the new version this looks fine to me for histogram data. For point data perhaps the following line `fs += sinus * inputY[inXIndex];` should really be calculating a y value that is midway between inXIndex and inXIndex+1 but I think the data will tend to be histogram data so suggest you don't worry.

PR #32947 (2021-11-24): Correct Integration range in PDFFourierTransform2
**Description of work.**
The integration range in PDFFourierTransform was too narrow, so a few fixes have allowed the full valid range of data to be included.
- Fixed problem where first_normal_index would consider 0 as not a normal value (when we want to only ignore Nan/Inf values).
- Added test for protected functions for determining the min/max indexes
- Cppcheck suppressions removed

- Also at line 385 there's an odd commented out section: https://github.com/mantidproject/mantid/pull/32947/files#diff-035333f5620c218e00e51a6236d91a87b7427912 [... cut]
closed 2021-11-24 by gemmaguest